ros2/rosidl #450: Used IDL generators may be different in two consecutive builds

ivanpauno, 2020-03-25:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10, Ubuntu 18.04
- Installation type:
  - From source, using a merged install path (`colcon build --merge-install`).
- Version or commit hash:
  - master
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
cd /path/to/workspace/src/ros2 && git clone https://github.com/ros2/rmw_dds_common/ && git checkout ivanpauno/first-implementation
cd /path/to/workspace/
colcon build --merge-install --packages-up-to rclpy
colcon build --merge-install --packages-select --cmake-force-configure rmw_dds_common
```

#### Expected behavior

It uses the same generators both times. i.e. the ones listed here: https://github.com/ros2/rmw_dds_common/blob/dc989e4f8f3fd2db997d6ba04aac6646b7e3acb3/rmw_dds_common_generators/package.xml#L17-L26

#### Actual behavior

The second time, also finds `rosidl_generator_py` and fails because it doesn't find `rosidl_typesupport_c`.
<details>
<summary>Failure message</summary>

```
CMake Error at /home/ivanpauno/ros2_ws/other_ws/merge_install_rebased_ros2_node_participant_ws/install/share/rosidl_cmake/cmake/rosidl_target_interfaces.cmake:44 (message):
  rosidl_target_interfaces() the second argument 'rmw_dds_common_interfaces'
  concatenated with the third argument 'rosidl_typesupport_c' using double
  underscores must be a valid target name
Call Stack (most recent call first):
  /home/ivanpauno/ros2_ws/other_ws/merge_install_rebased_ros2_node_participant_ws/install/share/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake:220 (rosidl_target_interfaces)
  /home/ivanpauno/ros2_ws/other_ws/merge_install_rebased_ros2_node_participant_ws/install/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:50 (include)
  /home/ivanpauno/ros2_ws/other_ws/merge_install_rebased_ros2_node_participant_ws/install/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:286 (ament_execute_extensions)
  CMakeLists.txt:27 (rosidl_generate_interfaces)
```

</details>

#### Additional information

I locally modified `ament_execute_extensions` to print the list of extensions found the second time, here the result:
```
rosidl_generator_c:rosidl_generator_c_generate_interfaces.cmake;rosidl_typesupport_connext_cpp:rosidl_typesupport_connext_cpp_generate_interfaces.cmake;rosidl_generator_cpp:rosidl_generator_c$
p_generate_interfaces.cmake;rosidl_typesupport_fastrtps_cpp:rosidl_typesupport_fastrtps_cpp_generate_interfaces.cmake;rosidl_typesupport_introspection_cpp:rosidl_typesupport_introspection_cpp_generate_in$
erfaces.cmake;rosidl_typesupport_introspection_c:rosidl_typesupport_introspection_c_generate_interfaces.cmake;rosidl_generator_py:rosidl_generator_py_generate_interfaces.cmake
```

The second time it finds `rosidl_generator_py` as an extension, but it doesn't find `rosidl_typesupport_c` --also already build and installed-- for some reason.
Ideally, it shouldn't find both, as they are not dependencies.
But it's strange that it finds one and not the other.

ivanpauno, 2020-03-25:
> The second time it finds rosidl_generator_py as an extension, but it doesn't find rosidl_typesupport_c --also already build and installed-- for some reason.
Ideally, it shouldn't find both, as they are not dependencies.
But it's strange that it finds one and not the other.

@dirk-thomas this seems to be the key of the problem.
Up to now, I have failed to see why it finds one and not the other the second time.

@hidmic FYI.

dirk-thomas, 2020-03-25:
> Expected behavior
> It uses the same generators both times. i.e. the ones listed here: https://github.com/ros2/rmw_dds_common/blob/dc989e4f8f3fd2db997d6ba04aac6646b7e3acb3/rmw_dds_common_generators/package.xml#L17-L26

Actually that is not the expected behavior. the CMake function `rosidl_generate_interfaces` will run all available generators - not only the ones defined in the manifest as a dependency - but all which are being `find_package()`-ed and therefore being registered as an extension in CMake.

> but it doesn't find rosidl_typesupport_c --also already build and installed-- for some reason.

While your package exports a dependency on `rosidl_typesupport_c` in the manifest I don't see it doing that in CMake anywhere?

ivanpauno, 2020-03-25:
> Actually that is not the expected behavior. the CMake function rosidl_generate_interfaces will run all available generators - not only the ones defined in the manifest as a dependency - but all which are being find_package()-ed and therefore being registered as an extension in CMake.

I find that pretty strange, because building the same workspace for a second time will have a different result.

> While your package exports a dependency on rosidl_typesupport_c in the manifest I don't see it doing that in CMake anywhere?

It does export a dependency on [rosidl_generator_cpp](https://github.com/ros2/rmw_dds_common/blob/dc989e4f8f3fd2db997d6ba04aac6646b7e3acb3/rmw_dds_common/CMakeLists.txt#L68), but not on `rosidl_typesupport_c`. `rosidl_typesupport_c` is not in the manifest either.

dirk-thomas, 2020-03-25:
It needs to pick up all available generators in order to support adding additional generators for other languages (which are not part of the default set).

ivanpauno, 2020-03-25:
> While your package exports a dependency on rosidl_typesupport_c in the manifest I don't see it doing that in CMake anywhere?

Where? I don't see it exports that dependency neither in `cmake` or in the manifest.

> It needs to pick up all available generators in order to support adding additional generators for other languages (which are not part of the default set).

I understand that, but if this new generator package uses a group membership on `rosidl_generator_packages`, it should be build before and included as a dependency of the package is generating the messages. Of course, getting all transitive dependencies --including group dependencies--, and just executing those extensions is probably hard.

dirk-thomas, 2020-03-25:
> Where? I don't see it exports that dependency neither in `cmake` or in the manifest.

My bad - only the C/C++ generators and introspection ts are in the manifest - bit not in CMake (https://github.com/ros2/rmw_dds_common/blob/dc989e4f8f3fd2db997d6ba04aac6646b7e3acb3/rmw_dds_common_generators/package.xml#L17-L21).

ivanpauno, 2020-03-27:
I fixed the problem in https://github.com/ros2/rmw_dds_common/pull/4/commits/1806af1bee5f5c70a217f2d2ec862af14c5fea47.
The issue was the packages I were exporting, as @dirk-thomas was probably guessing.

I was getting all `rosidl_generator_packages` that are registered as a ament resource and exporting them in a `CONFIG_EXTRA` file, where I should only be exporting the ones I need.

Closing as there's nothing to fix here.
Thanks @dirk-thomas for the help!

closed 2020-03-27 by ivanpauno