ros2/rosidl #451: Progress toward Quality Level 1

Blast545, 2020-03-25:
This issue tracks the progression of packages in `rosidl` that have been deemed necessary for Quality Level 1 and a 1.0 version level. It follows the outline described in [REP 2004](https://github.com/ros-infrastructure/rep/blob/c2ba03bac23b8f313e47820833fad000dbbe058b/rep-2004.rst). 

The packages slated for Quality Level 1 for ROS 2 Foxy are:
- rosidl_generator_c
- rosidl_generator_cpp
- rosidl_typesupport_interface

Pending:
- rosidl_adapter to be removed?
- rosidl_parser to be removed? 
(both based on code modifications being made to rosidl_parser)

Excluded for Quality Level 1 at this time:
- rosidl_cmake
- rosidl_typesupport_introspection_c
- rosidl_typesupport_introspection_cpp

# Progress common to all packages:
* [x] Version Policy
    - Follows ROS Core Quality Declaration
* [ ] Version >= 1.0.0
* [x] Change Control Process
    - Follows ROS Core

Documentation
* [x] Declared set of licenses
* [x] Copyright statement in each source file
    - Checked with amen_lint_common

Platform Support
* [x] Supports all tier 1 platforms

# Progress of specific packages
## rosidl_generator_c
Documentation
* [ ] Per-feature documentation.
* [ ] Per-item documentation in public API. 
* [ ] Quality Declaration document

Dependencies:
* [ ] Runtime "ROS" dependencies are level 1
  - rosidl_typesupport_interface
* [x] non-ROS dependencies are equivalent level 1
  - None

Testing:
* [ ] Must have system tests which cover all items in the "feature" documentation
* [ ] Must have system, integration, and/or unit tests which cover all of the public API
  - API and features to be defined, .h headers?
* [ ] Code Coverage:
  - * [x] Must have code coverage tracking for the package. [46%](https://ci.ros2.org/job/ci_linux_coverage/38/cobertura/src_ros2_rosidl_rosidl_generator_c_src/)
  - * [x] Must have and enforce a code coverage policy for new changes [Policy](https://index.ros.org/doc/ros2/Contributing/Developer-Guide/#coverage)
* [ ] Performance:
  - * [ ] Must have performance tests 
  - * [x] Must have a performance regression policy (i.e. blocking either changes or releases on unexpected performance regressions). -> [Policy](https://index.ros.org/doc/ros2/Contributing/Developer-Guide/#performance)
* [x] Linters and Static Analysis:
  - * [x] Must have a code style and enforce it
  - * [x] Must use static analysis tools where applicable: ament_lint_common enabled

## rosidl_generator_cpp
Documentation
* [ ] Per-feature documentation.
* [ ] Per-item documentation in public API. 
* [ ] Quality Declaration document

Dependencies:
* [ ] Runtime "ROS" dependencies are level 1
  - rosidl_parser (to be changed)
* [x] non-ROS dependencies are equivalent level 1
  - None

Testing:
* [ ] Must have system tests which cover all items in the "feature" documentation
* [ ] Must have system, integration, and/or unit tests which cover all of the public API
  - API and features to be defined, .hpp headers?
* [ ] Code Coverage:
  - * [x] Must have code coverage tracking for the package. [77%](https://ci.ros2.org/job/ci_linux_coverage/38/cobertura/src_ros2_rosidl_rosidl_generator_cpp_include_rosidl_generator_cpp/) 
  - * [x] Must have and enforce a code coverage policy for new changes [Policy](https://index.ros.org/doc/ros2/Contributing/Developer-Guide/#coverage)
* [ ] Performance:
  - * [ ] Must have performance tests 
  - * [x] Must have a performance regression policy (i.e. blocking either changes or releases on unexpected performance regressions). -> [Policy](https://index.ros.org/doc/ros2/Contributing/Developer-Guide/#performance)
* [x] Linters and Static Analysis:
  - * [x] Must have a code style and enforce it
  - * [x] Must use static analysis tools where applicable: ament_lint_common enabled

## rosidl_typesupport_interface
Documentation
* [ ] Per-feature documentation.
* [ ] Per-item documentation in public API. 
  - macro.h ?
* [ ] Quality Declaration document

Dependencies:
* [ ] Runtime "ROS" dependencies are level 1
  - None
* [x] non-ROS dependencies are equivalent level 1
  - None

Testing:
* [ ] Must have system tests which cover all items in the "feature" documentation
* [ ] Must have system, integration, and/or unit tests which cover all of the public API
  - This code defines just macros, use this as API?
* [x] Code Coverage:
  - * [x] Must have code coverage tracking for the package. Not needed
  - * [x] Must have and enforce a code coverage policy for new changes [Policy](https://index.ros.org/doc/ros2/Contributing/Developer-Guide/#coverage)
* [ ] Performance:
  - * [ ] Must have performance tests 
  - * [x] Must have a performance regression policy (i.e. blocking either changes or releases on unexpected performance regressions). -> [Policy](https://index.ros.org/doc/ros2/Contributing/Developer-Guide/#performance)
* [x] Linters and Static Analysis:
  - * [x] Must have a code style and enforce it
  - * [x] Must use static analysis tools where applicable: ament_lint_common enabled

chapulina, 2020-06-18:
@Blast545 , when the generators were split in 2 in https://github.com/ros2/rosidl/pull/442 we started focusing on the `rosidl_runtime_X` packages instead of the generators. Do you think you could add those to this issue? Their QDs are up-to-date for level 2 since #499 and #500, so this should be quick. Thanks!

Blast545, 2020-06-19:
@chapulina done. I added a couple of extra links to be more clear about the statements of this issue. Let me know if you think this is fine, to update other issues with this format.

chapulina, 2020-06-19:
Looks great, thanks!

chapulina, 2020-09-28:
All 3 packages are level 1! :tada:

closed 2020-09-28 by chapulina