ros2/rosidl #452: Failure to build under Windows in Github CI

rotu, 2020-03-30:
## Bug report

**Required Info:**

- Operating System:
  - Windows
- Installation type:
  - from source
- Version or commit hash:
  - dashing, eloquent
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Build from source with ros-tooling/action-ros-ci@0.0.14

#### Expected behavior

Build is successful, as in Master

#### Actual behavior

Build fails in `rosidl_generator_cpp` with output as below about a missing `rosidl_generator_c` library:

```
2020-03-30T20:27:03.2232931Z Starting >>> rosidl_generator_cpp
2020-03-30T20:27:16.3079994Z --- output: rosidl_generator_cpp
2020-03-30T20:27:16.3080769Z -- The C compiler identification is MSVC 19.25.28610.4
2020-03-30T20:27:16.3081244Z -- The CXX compiler identification is MSVC 19.25.28610.4
2020-03-30T20:27:16.3081706Z -- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Enterprise/VC/Tools/MSVC/14.25.28610/bin/Hostx64/x64/cl.exe
2020-03-30T20:27:16.3082181Z -- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Enterprise/VC/Tools/MSVC/14.25.28610/bin/Hostx64/x64/cl.exe -- works
2020-03-30T20:27:16.3082610Z -- Detecting C compiler ABI info
2020-03-30T20:27:16.3083004Z -- Detecting C compiler ABI info - done
2020-03-30T20:27:16.3083396Z -- Detecting C compile features
2020-03-30T20:27:16.3083790Z -- Detecting C compile features - done
2020-03-30T20:27:16.3084239Z -- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Enterprise/VC/Tools/MSVC/14.25.28610/bin/Hostx64/x64/cl.exe
2020-03-30T20:27:16.3084700Z -- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Enterprise/VC/Tools/MSVC/14.25.28610/bin/Hostx64/x64/cl.exe -- works
2020-03-30T20:27:16.3085133Z -- Detecting CXX compiler ABI info
2020-03-30T20:27:16.3085525Z -- Detecting CXX compiler ABI info - done
2020-03-30T20:27:16.3085921Z -- Detecting CXX compile features
2020-03-30T20:27:16.3086313Z -- Detecting CXX compile features - done
2020-03-30T20:27:16.3086782Z -- Found ament_cmake: 0.8.2 (D:/a/rmw_cyclonedds/rmw_cyclonedds/ros_ws/install/ament_cmake/share/ament_cmake/cmake)
2020-03-30T20:27:16.3087231Z -- Found PythonInterp: C:/python37/python.exe (found suitable version "3.7.6", minimum required is "3") 
2020-03-30T20:27:16.3087690Z -- Using PYTHON_EXECUTABLE: C:/python37/python.exe
2020-03-30T20:27:16.3088007Z -- Override CMake install command with custom implementation using symlinks instead of copying resources
2020-03-30T20:27:16.3088310Z -- Found ament_lint_auto: 0.8.1 (D:/a/rmw_cyclonedds/rmw_cyclonedds/ros_ws/install/ament_lint_auto/share/ament_lint_auto/cmake)
2020-03-30T20:27:16.3088920Z -- Found rosidl_generator_c: 0.8.2 (D:/a/rmw_cyclonedds/rmw_cyclonedds/ros_ws/install/rosidl_generator_c/share/rosidl_generator_c/cmake)
2020-03-30T20:27:16.3089233Z -- Found rosidl_adapter: 0.8.2 (D:/a/rmw_cyclonedds/rmw_cyclonedds/ros_ws/install/rosidl_adapter/share/rosidl_adapter/cmake)
2020-03-30T20:27:16.3089557Z CMake Error at D:/a/rmw_cyclonedds/rmw_cyclonedds/ros_ws/install/rosidl_generator_c/share/rosidl_generator_c/cmake/ament_cmake_export_libraries-extras.cmake:48 (message):
2020-03-30T20:27:16.3089850Z   Package 'rosidl_generator_c' exports the library 'rosidl_generator_c' which
2020-03-30T20:27:16.3090100Z   couldn't be found
2020-03-30T20:27:16.3090336Z Call Stack (most recent call first):
2020-03-30T20:27:16.3090777Z   D:/a/rmw_cyclonedds/rmw_cyclonedds/ros_ws/install/rosidl_generator_c/share/rosidl_generator_c/cmake/rosidl_generator_cConfig.cmake:38 (include)
2020-03-30T20:27:16.3091095Z   CMakeLists.txt:19 (find_package)
2020-03-30T20:27:16.3091284Z 
2020-03-30T20:27:16.3091471Z 
2020-03-30T20:27:16.3091708Z -- Configuring incomplete, errors occurred!
2020-03-30T20:27:16.3092027Z See also "D:/a/rmw_cyclonedds/rmw_cyclonedds/ros_ws/build/rosidl_generator_cpp/CMakeFiles/CMakeOutput.log".
2020-03-30T20:27:16.3092302Z ---
2020-03-30T20:27:16.3092663Z --- stderr: rosidl_generator_cpp
2020-03-30T20:27:16.3092986Z CMake Error at D:/a/rmw_cyclonedds/rmw_cyclonedds/ros_ws/install/rosidl_generator_c/share/rosidl_generator_c/cmake/ament_cmake_export_libraries-extras.cmake:48 (message):
2020-03-30T20:27:16.3093302Z   Package 'rosidl_generator_c' exports the library 'rosidl_generator_c' which
2020-03-30T20:27:16.3093570Z   couldn't be found
2020-03-30T20:27:16.3093814Z Call Stack (most recent call first):
2020-03-30T20:27:16.3094117Z   D:/a/rmw_cyclonedds/rmw_cyclonedds/ros_ws/install/rosidl_generator_c/share/rosidl_generator_c/cmake/rosidl_generator_cConfig.cmake:38 (include)
2020-03-30T20:27:16.3094401Z   CMakeLists.txt:19 (find_package)
2020-03-30T20:27:16.3094588Z 
2020-03-30T20:27:16.3094783Z 
2020-03-30T20:27:16.3095018Z ---
2020-03-30T20:27:16.3095269Z Failed   <<< rosidl_generator_cpp	[ Exited with code 1 ]
```

https://github.com/RoverRobotics-forks/rmw_cyclonedds/runs/546682290

#### Additional information

It does seem odd that the error is in `rosidl_generator_cpp`, although the library in question is 'rosidl_generator_c'
This may have to do with this odd-looking hack, which I do not understand:
https://github.com/ros2/rosidl/blob/eloquent/rosidl_generator_c/CMakeLists.txt#L60

rotu, 2020-03-30:
FYI @ros2/tooling_working_group

dirk-thomas, 2020-03-30:
I would suggest to create a ticket in the repository providing the GitHub CI actions instead. This package builds just fine on Windows on ci.ros2.org so it is unlikely that the problem you observed is related to the code in this repository.

rotu, 2020-03-30:
Thanks, @dirk-thomas. I'm building a comparable build here [image] and I also see some rosdep warnings which I'll look into as well. It may be a problem in tooling instead. I suspect the issue has something to do with both some unusual things in the CMake file as well as rare build settings (e.g. `--symlink-install` on Windows).

rotu, 2020-03-30:
Update:
without `--symlink-install`, it gets past this part of the build. [image]
Seems like the bug fixed in https://github.com/ament/ament_cmake/pull/219

rotu, 2020-04-06:
@dirk-thomas, @nuclearsandwich, are you waiting on my action?

dirk-thomas, 2020-04-06:
Both PRs have a reviewer associated with it. Those two people are responsible to review / merge the PRs.