ros2/rosidl #469: Build fails without rosidl_typesupport_fastrtps

alsora, 2020-04-21:
## Bug report

**Required Info:**

- Operating System:
  - ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - master

Hi, since some of the changes related to https://github.com/ros2/rosidl/issues/447 got merged, we found a bug when compiling ROS 2 from sources.
I think the actual bug may have been introduced with  https://github.com/ros2/rosidl/pull/461
If you build ignoring FastRTPS and Connext then the build fails

```
touch \
  src/eProsima/COLCON_IGNORE \
  src/ros2/rmw_connext/COLCON_IGNORE \
  src/ros2/rmw_fastrtps/COLCON_IGNORE \
  src/ros2/rosidl_typesupport_connext/COLCON_IGNORE \
  src/ros2/rosidl_typesupport_fastrtps/COLCON_IGNORE
```

```
Starting >>> builtin_interfaces
--- stderr: builtin_interfaces                                      
/root/ros2_ws/build/builtin_interfaces/rosidl_typesupport_c/builtin_interfaces/msg/duration__type_support.cpp:10:10: fatal error: builtin_interfaces/msg/detail/duration__rosidl_typesupport_introspection_c.h: No such file or directory
 #include "builtin_interfaces/msg/detail/duration__rosidl_typesupport_introspection_c.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

Restoring `rosidl_typesupport_fastrtps` fixes it.
This was not required before this change, when we were able to compile only for Fast-RTPS or only for CycloneDDS.

mauropasse, 2020-04-21:
I have a horrible temporary solution:
1st, apply this patch:

```
diff --git a/rosidl_typesupport_introspection_c/cmake/rosidl_typesupport_introspection_c_generate_interfaces.cmake b/rosidl_typesupport_introspection_c/cmake/rosidl_typesupport_introspection_c_generate_interfaces.cmake
index d3397ad..aa74677 100644
--- a/rosidl_typesupport_introspection_c/cmake/rosidl_typesupport_introspection_c_generate_interfaces.cmake
+++ b/rosidl_typesupport_introspection_c/cmake/rosidl_typesupport_introspection_c_generate_interfaces.cmake
@@ -22,7 +22,7 @@ foreach(_abs_idl_file ${rosidl_generate_interfaces_ABS_IDL_FILES})
   get_filename_component(_idl_name "${_abs_idl_file}" NAME_WE)
   string_camel_case_to_lower_case_underscore("${_idl_name}" _header_name)
   list(APPEND _generated_header_files
-    "${_output_path}/${_parent_folder}/detail/${_header_name}__rosidl_typesupport_introspection_c.h")
+    "${_output_path}/${_parent_folder}/${_header_name}__rosidl_typesupport_introspection_c.h")
   list(APPEND _generated_source_files
     "${_output_path}/${_parent_folder}/${_header_name}__type_support.c")
 endforeach()
diff --git a/rosidl_typesupport_introspection_c/rosidl_typesupport_introspection_c/__init__.py b/rosidl_typesupport_introspection_c/rosidl_typesupport_introspection_c/__init__.py
index f18c0f3..6affa9e 100644
--- a/rosidl_typesupport_introspection_c/rosidl_typesupport_introspection_c/__init__.py
+++ b/rosidl_typesupport_introspection_c/rosidl_typesupport_introspection_c/__init__.py
@@ -18,7 +18,7 @@ from rosidl_cmake import generate_files
 def generate_c(generator_arguments_file):
     mapping = {
         'idl__rosidl_typesupport_introspection_c.h.em':
-        'detail/%s__rosidl_typesupport_introspection_c.h',
+        '%s__rosidl_typesupport_introspection_c.h',
         'idl__type_support.c.em': '%s__type_support.c',
     }
     generate_files(generator_arguments_file, mapping)
diff --git a/rosidl_typesupport_introspection_cpp/cmake/rosidl_typesupport_introspection_cpp_generate_interfaces.cmake b/rosidl_typesupport_introspection_cpp/cmake/rosidl_typesupport_introspection_cpp_generate_interfaces.cmake
index fcde74d..6f65469 100644
--- a/rosidl_typesupport_introspection_cpp/cmake/rosidl_typesupport_introspection_cpp_generate_interfaces.cmake
+++ b/rosidl_typesupport_introspection_cpp/cmake/rosidl_typesupport_introspection_cpp_generate_interfaces.cmake
@@ -22,7 +22,7 @@ foreach(_abs_idl_file ${rosidl_generate_interfaces_ABS_IDL_FILES})
   get_filename_component(_idl_name "${_abs_idl_file}" NAME_WE)
   string_camel_case_to_lower_case_underscore("${_idl_name}" _header_name)
   list(APPEND _generated_header_files
-    "${_output_path}/${_parent_folder}/detail/${_header_name}__rosidl_typesupport_introspection_cpp.hpp")
+    "${_output_path}/${_parent_folder}/${_header_name}__rosidl_typesupport_introspection_cpp.hpp")
   list(APPEND _generated_source_files
     "${_output_path}/${_parent_folder}/${_header_name}__type_support.cpp")
 endforeach()
diff --git a/rosidl_typesupport_introspection_cpp/rosidl_typesupport_introspection_cpp/__init__.py b/rosidl_typesupport_introspection_cpp/rosidl_typesupport_introspection_cpp/__init__.py
index 9713ead..a3d6e6d 100644
--- a/rosidl_typesupport_introspection_cpp/rosidl_typesupport_introspection_cpp/__init__.py
+++ b/rosidl_typesupport_introspection_cpp/rosidl_typesupport_introspection_cpp/__init__.py
@@ -18,7 +18,7 @@ from rosidl_cmake import generate_files
 def generate_cpp(generator_arguments_file):
     mapping = {
         'idl__rosidl_typesupport_introspection_cpp.hpp.em':
-        'detail/%s__rosidl_typesupport_introspection_cpp.hpp',
+        '%s__rosidl_typesupport_introspection_cpp.hpp',
         'idl__type_support.cpp.em': '%s__type_support.cpp',
     }
     generate_files(generator_arguments_file, mapping)
```
2nd: Compile (it will fail). After it fails, revert the patch, compile again, it works. xD

dirk-thomas, 2020-04-21:
You are probably mixing the latest state of this repository with an outdated version of other repos. Please make sure to get the latest state from all repos.

mauropasse, 2020-04-21:
I can confirm I'm using the latest `ros2.repos` and it fails when ignoring FastRTPS and Connext

dirk-thomas, 2020-04-21:
Please provide a complete list of repos hashes to confirm (`vcs export --exact`).

dirk-thomas, 2020-04-21:
Nvm, I think http://build.ros2.org/view/Fci/job/Fci__nightly-cyclonedds_ubuntu_focal_amd64/86/ confirms the issue.

closed 2020-04-21 by dirk-thomas

alsora, 2020-04-22:
Confirm that https://github.com/ros2/rosidl/pull/470 fixes the issue

PR #470 (2020-04-21): fix build with single introspection typesupport
Fixes #469.

CI with only CycloneDDS and up to `demo_nodes_cpp` / `demo_nodes_py`:
Before: [image]
After: [image]

Same but with all three RMW: [image]