ros2/rosidl #479: Failed Build - Debian - rosidl_generator_traits::is_message

mrlowalowa, 2020-04-30:
## Bug report

**Required Info:**

- Operating System:
  - Linux debian 4.9.0-11-rt-amd64 #1 SMP PREEMPT RT Debian 4.9.189-3+deb9u2 (2019-11-11) x86_64 GNU/Linux
- Installation type:
  -  from source 
- Version or commit hash:
  - ros2-eloquent latest
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  -  N/A 

#### Steps to reproduce issue

```
https://index.ros.org/doc/ros2/Installation/Eloquent/Linux-Development-Setup/
--> step: colcon build
https://github.com/ros2/ros2/issues/470#issuecomment-371141641 (cmake from source)
```

#### Expected behavior
Build ros eloquent from Source on debian. Already build cmake 10 from source for https support in libcurl.
#### Actual behavior
```Starting >>> rosidl_generator_cpp
[Processing: rosidl_generator_cpp, rviz_assimp_vendor, rviz_ogre_vendor]
[Processing: rosidl_generator_cpp, rviz_assimp_vendor, rviz_ogre_vendor]
--- stderr: rosidl_generator_cpp
/home/pt1/gits/ros2_eloquent/src/ros2/rosidl/rosidl_generator_cpp/test/test_traits.cpp:32:33: error: specialization of
template<class T> struct rosidl_generator_traits::is_message’ in different namespace [-fpermissive]
 struct rosidl_generator_traits::is_message<Message2>: std::true_type {};
                                 ^~~~~~~~~~~~~~~~~~~~
In file included from /home/pt1/gits/ros2_eloquent/build/rosidl_generator_cpp/rosidl_generator_cpp/rosidl_generator_cpp/msg/empty__traits.hpp:9:0,
                 from /home/pt1/gits/ros2_eloquent/build/rosidl_generator_cpp/rosidl_generator_cpp/rosidl_generator_cpp/msg/empty.hpp:8,
                 from /home/pt1/gits/ros2_eloquent/src/ros2/rosidl/rosidl_generator_cpp/test/test_traits.cpp:16:
/home/pt1/gits/ros2_eloquent/src/ros2/rosidl/rosidl_generator_cpp/include/rosidl_generator_cpp/traits.hpp:34:8: error:   from definition of ‘template<class T> struct rosidl_generator_traits::is_message’ [-fpermissive]
 struct is_message : std::false_type {};
        ^~~~~~~~~~
make[2]: *** [CMakeFiles/test_traits.dir/test/test_traits.cpp.o] Error 1
make[1]: *** [CMakeFiles/test_traits.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< rosidl_generator_cpp       [ Exited with code 2 ]
Aborted  <<< rviz_assimp_vendor
Aborted  <<< rviz_ogre_vendor

Summary: 94 packages finished [1h 11min 17s]
  1 package failed: rosidl_generator_cpp
  2 packages aborted: rviz_assimp_vendor rviz_ogre_vendor
  1 package had stderr output: rosidl_generator_cpp
  201 packages not processed```
#### Additional information

DentOpt, 2020-05-01:
Exactly  same issue for ros2 Foxy on tinkerboard armv7l GNU/Linux

clalancette, 2020-05-01:
Debian is one of the Tier 3 platforms for [Eloquent](https://index.ros.org/doc/ros2/Releases/Release-Eloquent-Elusor/) and [Foxy](https://index.ros.org/doc/ros2/Releases/Release-Foxy-Fitzroy/).  That means that while we aim to support it, its primarily a community effort to keep it running (see https://www.ros.org/reps/rep-2000.html#support-tiers for details).

Thus, we would happily review any pull requests to fix things on Debian (9 or 10), but we probably will not have the time to look into it ourselves.

DentOpt, 2020-05-05:
While compiling for Debian armhf (TinkerOS)=-> 
I could narrow down the error currently to 

`colcon build --packages-select sros2 --symlink-install --event-handlers console_cohesion+
Starting >>> sros2   
Traceback (most recent call last):8s]
  File "<string>", line 1, in <module>
  File "/usr/lib/python3.5/distutils/core.py", line 216, in run_setup
    exec(f.read(), g, l)
  File "<string>", line 17, in <module>
  File "<string>", line 10, in package_files
NameError: name 'os' is not defined
--- output: sros2                    
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_core/executor/__init__.py", line 91, in __call__
    rc = await self.task(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_core/task/__init__.py", line 93, in __call__
    return await task_method(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_ros/task/ament_python/build.py", line 51, in build
    setup_py_data = get_setup_data(self.context.pkg, env)
  File "/usr/lib/python3/dist-packages/colcon_core/task/python/__init__.py", line 20, in get_setup_data
    return dict(pkg.metadata[key](env))
  File "/usr/lib/python3/dist-packages/colcon_ros/package_identification/ros.py", line 130, in getter
    desc.path / 'setup.py', env=env)
  File "/usr/lib/python3/dist-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 265, in get_setup_information
    setup_py, env=env)
  File "/usr/lib/python3/dist-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 303, in _get_setup_information
    cwd=os.path.abspath(str(setup_py.parent)), check=True, env=env)
  File "/usr/lib/python3.5/subprocess.py", line 398, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-c', "import sys;from distutils.core import run_setup;dist = run_setup(    'setup.py', script_args=('--dry-run',), stop_after='config');skip_keys = ('cmdclass', 'distclass', 'ext_modules', 'metadata');data = {    key: value for key, value in dist.__dict__.items()     if (        not key.startswith('_') and         not callable(value) and         key not in skip_keys and         key not in dist.display_option_names    )};data['metadata'] = {    k: v for k, v in dist.metadata.__dict__.items()     if k not in ('license_files', 'provides_extras')};sys.stdout.buffer.write(repr(data).encode('utf-8'))"]' returned non-zero exit status 1
---
--- stderr: sros2
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_core/executor/__init__.py", line 91, in __call__
    rc = await self.task(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_core/task/__init__.py", line 93, in __call__
    return await task_method(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_ros/task/ament_python/build.py", line 51, in build
    setup_py_data = get_setup_data(self.context.pkg, env)
  File "/usr/lib/python3/dist-packages/colcon_core/task/python/__init__.py", line 20, in get_setup_data
    return dict(pkg.metadata[key](env))
  File "/usr/lib/python3/dist-packages/colcon_ros/package_identification/ros.py", line 130, in getter
    desc.path / 'setup.py', env=env)
  File "/usr/lib/python3/dist-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 265, in get_setup_information
    setup_py, env=env)
  File "/usr/lib/python3/dist-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 303, in _get_setup_information
    cwd=os.path.abspath(str(setup_py.parent)), check=True, env=env)
  File "/usr/lib/python3.5/subprocess.py", line 398, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-c', "import sys;from distutils.core import run_setup;dist = run_setup(    'setup.py', script_args=('--dry-run',), stop_after='config');skip_keys = ('cmdclass', 'distclass', 'ext_modules', 'metadata');data = {    key: value for key, value in dist.__dict__.items()     if (        not key.startswith('_') and         not callable(value) and         key not in skip_keys and         key not in dist.display_option_names    )};data['metadata'] = {    k: v for k, v in dist.metadata.__dict__.items()     if k not in ('license_files', 'provides_extras')};sys.stdout.buffer.write(repr(data).encode('utf-8'))"]' returned non-zero exit status 1
---
Failed   <<< sros2	[ Exited with code 1 ]

Summary: 0 packages finished [10.2s]
  1 package failed: sros2
  1 package had stderr output: sros2
Command '['/usr/bin/python3', '-c', "import sys;from distutils.core import run_setup;dist = run_setup(    'setup.py', script_args=('--dry-run',), stop_after='config');skip_keys = ('cmdclass', 'distclass', 'ext_modules', 'metadata');data = {    key: value for key, value in dist.__dict__.items()     if (        not key.startswith('_') and         not callable(value) and         key not in skip_keys and         key not in dist.display_option_names    )};data['metadata'] = {    k: v for k, v in dist.metadata.__dict__.items()     if k not in ('license_files', 'provides_extras')};sys.stdout.buffer.write(repr(data).encode('utf-8'))"]' returned non-zero exit status 1
`

The strange thing is that os is not recognized even while "import os" is given in "/usr/lib/python3.5/distutils/core.py". 

Did anyone face this issue?

In order to track this error down, is there any documentation on 
with which parameters I would need to run 
"/usr/lib/python3/dist-packages/colcon_core/executor/__init__.py"
in order to reproduce the behavior of 
"colcon build --packages-select sros2 --symlink-install --event-handlers console_cohesion+"

shoeffner, 2020-08-05:
I had the same issue on Debian 9.

This seems to be a [bug in gcc](https://stackoverflow.com/a/25594741), and despite it saying the target milestone was 7 it still occurred when I installed gcc 7.5. Anyways, the [original bug report](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=56480) states that:

```cpp
namespace Foo 
{
    template <typename T>
    struct Meow
    {
    };
}

template <>
struct Foo::Meow<int>
{
};
```

failed, while 

> The code is accepted if a declaration (without definition) of the specialization is placed in the namespace.

So changing 

https://github.com/ros2/rosidl/blob/76c536eb35b15076c619f835bea3561e2c6fc6a3/rosidl_generator_cpp/test/test_traits.cpp#L31-L32

to

```cpp
namespace rosidl_generator_traits
{

template<>
struct is_message<Message2>: std::true_type {};

}
```

Resolves this issue. To keep the current functionality, one option would be to check the compiler, something along these lines (untested):

```cpp
#if defined(__GNUC__)
  #if __GNUC__ > 7
template<>
struct rosidl_generator_traits::is_message<Message2>: std::true_type {};
  #else
namespace rosidl_generator_traits
{

template<>
struct is_message<Message2>: std::true_type {};

}
  #endif
#endif
```

But this might get messy over time. 

In any case, here's the patch I mentioned above:
```patch
From 5f3845f0e915fcc26330cd86061dc79db11aa1b0 Mon Sep 17 00:00:00 2001
From: =?UTF-8?q?Sebastian=20H=C3=B6ffner?= <info@sebastian-hoeffner.de>
Date: Wed, 5 Aug 2020 18:32:07 +0200
Subject: [PATCH] Declaring is_message in namespace rosidl_generator_traits.

Resolves issues with older GCC versions, as described in https://gcc.gnu.org/bugzilla/show_bug.cgi?id=56480

Resolves #479
---
 rosidl_generator_cpp/test/test_traits.cpp | 7 ++++++-
 1 file changed, 6 insertions(+), 1 deletion(-)

diff --git a/rosidl_generator_cpp/test/test_traits.cpp b/rosidl_generator_cpp/test/test_traits.cpp
index eb3d5fc..dc9dcc6 100644
--- a/rosidl_generator_cpp/test/test_traits.cpp
+++ b/rosidl_generator_cpp/test/test_traits.cpp
@@ -28,8 +28,13 @@ struct Message {};
 // Empty testing struct, with template instantiation
 struct Message2 {};
 
+namespace rosidl_generator_traits
+{
+
 template<>
-struct rosidl_generator_traits::is_message<Message2>: std::true_type {};
+struct is_message<Message2>: std::true_type {};
+
+}
 
 TEST(Test_rosidl_generator_traits, is_message) {
   // A message is not a service
-- 
2.11.0
```

dirk-thomas, 2020-08-05:
> In any case, here's the patch I mentioned above:

@shoeffner If you would like this proposed patch to be reviewed and considered please create a pull request.

PR #512 (2020-08-11): Declaring is_message in namespace rosidl_generator_traits.
Resolves issues with older GCC versions, as described in https://gcc.gnu.org/bugzilla/show_bug.cgi?id=56480.

Resolves #479.

#479 seems to be a [bug in gcc](https://stackoverflow.com/a/25594741). The [original bug report](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=56480) states that:

```cpp
namespace Foo 
{
    template <typename T>
    struct Meow
    {
    };
}

template <>
struct Foo::Meow<int>
{
};
```

failed, while 

> The code is accepted if a declaration (without definition) of the specialization is placed in the namespace. [... cut]
commit bbc154ba (2020-08-11): Declaring is_message in namespace rosidl_generator_traits. (#512)

Resolves issues with older GCC versions, as described in https://gcc.gnu.org/bugzilla/show_bug.cgi?id=56480

Resolves #479
closed 2020-08-11 by dirk-thomas