ros2/rosidl #486: BoundedVector misuses InputIterators

jwakely, 2020-05-12:
Input iterators are single pass, meaning once you've incremented them, they're gone. That means functions like [this](https://github.com/ros2/rosidl/blob/afff057f283959d33da065349eb971b8b3653170/rosidl_runtime_cpp/include/rosidl_runtime_cpp/bounded_vector.hpp#L283) are broken for input iterators. The call to `distance(first, last)` will consume the entire range, and then `assign(first, last)` will do nothing.

```
#include <sstream>
#include <iterator>
#include <cassert>
#include <rosidl_runtime_cpp/bounded_vector.hpp>
int main()
{
  std::istringstream s("1 2 3");
  BoundedVector<int> b;
  std::istream_iterator<int> iter(s), end;
  b.assign(iter, s);
  assert(b.size() == 3);
}
```
You should either change that function to only accept forward iterators, or use tag dispatching to select different functions depending on `std::iterator_traits<InputIterator>::iterator_category`. For input iterators you could just do something like:

```
Base::clear();
while (first != last)
  push_back(*first++);
```
For forward iterators you can use `std::distance(first, last)` safely.

jwakely, 2020-05-12:
Alternatively, do what you do in the constructor taking a pair of iterators, and do the assignment first and then check:

```
Base::assign(first, last);
if (size() > UpperBound) {
  clear();
  throw std::length_error("...");
}
```
This doesn't offer the strong exception-safety guarantee though, and potentially allocates a lot of memory only to throw it away again.

hidmic, 2020-05-14:
@jwakely would you willing to contribute a fix?

jwakely, 2020-05-14:
See https://github.com/jwakely/rosidl/commit/2a5ebbee395bd8b7b113c9a3f0b36ee27ea43bc1 for a solution. That branch also contains the commits in pull request #487 so I haven't created a pull request for the fix for this issue. I'll do so after a decision is made about merging #487 or not (separating the fix for this issue from the other commits if needed).

jwakely, 2020-05-20:
Now that pull req #487 has been merged I've rebased the fix for this issue and created pull req #493.

PR #493 (2020-06-19): Fix misuses of input iterators in BoundedVector
Input iterators are single pass, so calling std::distance(first, last)
will consume the input and leave [first,last) as an empty range.

This change makes assign(InputIterator, InputIterator) and
insert(const_iterator, InputIterator, InputIterator) dispatch to
overloaded functions that choose the right overload based on the
iterator category. For forward iterators the original implementation is
used. For input iterators assign will construct a new vector and if that
doesn't throw, use swap to replace the contents of *this. For insert
each value i [... cut]
commit 853481b6 (2020-06-19): Fix misuses of input iterators in BoundedVector (#493)

* Fix misuses of input iterators in BoundedVector

Input iterators are single pass, so calling std::distance(first, last)
will consume the input and leave [first,last) as an empty range.

This change makes assign(InputIterator, InputIterator) and
insert(const_iterator, InputIterator, InputIterator) dispatch to
overloaded functions that choose the right overload based on the
iterator category. For forward iterators the original implementation is
used. For input iterators assign will construct a new vector and if that
doesn't throw, use swa [... cut]
closed 2020-06-19 by hidmic