ros2/rosidl #491: python scripts installed in a nonstandard way

rotu, 2020-05-19:
The rosidl entry points `rosidl_generator_c` and `rosidl_generator_cpp` seem to be executable python scripts but for some reason have no extension and are installed with CMake.

https://github.com/ros2/rosidl/blob/master/rosidl_generator_c/bin/rosidl_generator_c
https://github.com/ros2/rosidl/blob/master/rosidl_generator_cpp/bin/rosidl_generator_cpp
https://github.com/ros2/rosidl/blob/master/rosidl_parser/bin/idl2png
https://github.com/ros2/rosidl/blob/master/rosidl_typesupport_introspection_c/bin/rosidl_typesupport_introspection_c
https://github.com/ros2/rosidl/blob/master/rosidl_typesupport_introspection_cpp/bin/rosidl_typesupport_introspection_cpp

It seems odd to have done it this way versus using the standard Python way: `setup.py` with a `console_scripts` entry point.

One weird side effect is that missing dependencies show up as import errors (notice the missing lark seems to be *caused by* rather than causing the rosidl_generator_c failure):
```
--- stderr: rosidl_generator_c                                                                                                    
Traceback (most recent call last):
  File "/opt/ros/master/src/ros2/rosidl/rosidl_generator_c/bin/rosidl_generator_c", line 8, in <module>
    from rosidl_generator_c import generate_c
ModuleNotFoundError: No module named 'rosidl_generator_c'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/ros/master/src/ros2/rosidl/rosidl_generator_c/bin/rosidl_generator_c", line 20, in <module>
    rosidl_generator_c = loader.load_module()
  File "<builtin>/frozen importlib._bootstrap_external", line 412, in _check_name_wrapper
  File "<builtin>/frozen importlib._bootstrap_external", line 836, in load_module
  File "<builtin>/frozen importlib._bootstrap_external", line 695, in load_module
  File "<frozen importlib._bootstrap>", line 274, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 693, in _load
  File "<frozen importlib._bootstrap>", line 674, in _load_unlocked
  File "<builtin>/frozen importlib._bootstrap_external", line 691, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/opt/ros/master/src/ros2/rosidl/rosidl_generator_c/rosidl_generator_c/__init__.py", line 15, in <module>
    from rosidl_cmake import convert_camel_case_to_lower_case_underscore
  File "/opt/ros/master/install/site-packages/rosidl_cmake/__init__.py", line 24, in <module>
    from rosidl_parser.parser import parse_idl_file
  File "/opt/ros/master/install/site-packages/rosidl_parser/parser.py", line 20, in <module>
    from lark import Lark
ModuleNotFoundError: No module named 'lark'
make[2]: *** [CMakeFiles/rosidl_generator_c_interfaces__rosidl_generator_c.dir/build.make:85: rosidl_generator_c/rosidl_generator_c/msg/arrays.h] Error 1
make[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/rosidl_generator_c_interfaces__rosidl_generator_c.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rosidl_generator_c	[ Exited with code 2 ]
```

rotu, 2020-05-19:
A similar issue is found in some other packages:

`rosidl_typesupport`:
* https://github.com/ros2/rosidl_typesupport/tree/master/rosidl_typesupport_c/bin
* https://github.com/ros2/rosidl_typesupport/tree/master/rosidl_typesupport_cpp/bin

`rosidl_dds`:
* https://github.com/ros2/rosidl_dds/tree/master/rosidl_generator_dds_idl/bin

`ros1_bridge`:
* https://github.com/ros2/ros1_bridge/tree/master/bin

dirk-thomas, 2020-05-19:
> The rosidl entry points `rosidl_generator_c` and `rosidl_generator_cpp` seem to be executable python scripts but for some reason have no extension and are installed with CMake.

A package can only have a single build system. All referenced packages use CMake - not Python setuptools. Therefore the scripts are being installed using the CMake function `install()`. 

> It seems odd to have done it this way versus using the standard Python way: `setup.py` with a `console_scripts` entry point.

Since the referenced package heavily use CMake, e.g. registering `ament_cmake` extensions, being `find_package()`-ed by downstream packages, it doesn't look feasible to convert them into Python packages.

So until a CMake function is being implemented which invoked Python setuptools (either directly or emulating the entry point logic) a CMake package can't have a `setup.py` file to create actually compiled Python entry points. This is ticketed in ament/ament_cmake#213.

Therefore I will close this ticket since it is not actionable atm. Once The above CMake function has been implemented it will be re-evaluated where it can be used.

closed 2020-05-19 by dirk-thomas

rotu, 2020-05-19:
Please reopen. There are still actionable issues here:

1. Why does an import hack exist in [`rosidl_generator_c`](https://github.com/ros2/rosidl/blob/87132fc2114c6e39f7686912ca9e590b15a2fd72/rosidl_generator_c/bin/rosidl_generator_c#L7-L21) and not in [`rosidl_typesupport_introspection_c`](https://github.com/ros2/rosidl/blob/master/rosidl_typesupport_introspection_c/bin/rosidl_typesupport_introspection_c)? This code seems questionable.

2. As per https://github.com/ament/ament_cmake/issues/213, there is an actionable solution: split it into a separate package.

3. If that is not satisfactory, https://github.com/ament/ament_cmake/issues/213, then this issue should remain open until upstream changes make it possible to resolve this issue satisfactorily. It would be more appropriate to assign some sort of tag like "pending upstream" rather than closing the issue.

dirk-thomas, 2020-05-19:
> Why does an import hack exist in rosidl_generator_c and not in rosidl_typesupport_introspection_c? This code seems questionable.

This question has not been mentioned above and seems unrelated to the title of this ticket. The `rosidl_generator_c` package uses the script in its own test (see https://github.com/ros2/rosidl/blob/87132fc2114c6e39f7686912ca9e590b15a2fd72/rosidl_generator_c/CMakeLists.txt#L35-L50) and needed this logic on Windows as described in the comment in the code. The `rosidl_typesupport_introspection_c` doesn't do that and therefore doesn't need the logic.

> As per ament/ament_cmake#213, there is an actionable solution: split it into a separate package.

While splitting the package would be a possible workaround it is nothing we are planning to do. Instead we want to utilize the to-be-implemented CMake function to allow Python entry points in CMake packages.

> then this issue should remain open until upstream changes make it possible to resolve this issue satisfactorily. It would be more appropriate to assign some sort of tag like "pending upstream" rather than closing the issue.

I added a note to the upstream ticket (see https://github.com/ament/ament_cmake/issues/213#issuecomment-631092262) which should be enough to follow up on this. A central note in that ticket will also apply to other packages and isn't limited to this repo like this ticket is. Please feel free to comment on the `ament_cmake` ticket to collect locations you think should be considered to be updated after the feature has been implemented.

rotu, 2020-05-19:
> This question has not been mentioned above and seems unrelated to the title of this ticket.

Because it seemed to be a weird workaround that would be resolved by installing python scripts in a standard way. Makes sense that this is for the testing code.

> Instead we want to utilize the to-be-implemented CMake function to allow Python entry points in CMake packages.

:+1: Although there's a little bit more to it than that. The `rosidl_generator_c` Python package is used by both `rosidl_generator_c` and `rosidl_typesupport_introspection_c`. Adding support for Python entry points to CMake is not sufficient as there is still the supporting code to install - we also should either (1) broaden the upstream scope to add support for Python packages, not just modules (2) Add a `setup.py` within this `ament_cmake` package and invoke `python setup.py`, `pip` or `colcon` on that Python subpackage (3) extract the Python implementation details into a separate package.

https://github.com/ros2/rosidl/blob/master/rosidl_generator_c/bin/rosidl_generator_c#L8 and https://github.com/ros2/rosidl/blob/master/rosidl_typesupport_introspection_c/resource/msg__type_support.c.em#L4

> I added a note to the upstream ticket (see [ament/ament_cmake#213 (comment)](https://github.com/ament/ament_cmake/issues/213#issuecomment-631092262)) which should be enough to follow up on this. A central note in that ticket will also apply to other packages and isn't limited to this repo like this ticket is. Please feel free to comment on the `ament_cmake` ticket to collect locations you think should be considered to be updated after the feature has been implemented.

That's sensible. I still have trouble wrapping my head around where to report tickets since ROS spans a massive number of repos. I kinda tend to lean toward "the issue goes in whatever repo is functionally affected", with the idea that "issues are for problems, PRs are for action". But I can see the logic that the issue info should stay closest to the code that first needs changing.

In a perfect world, ROS or GitHub would have a combined ticketing system that didn't couple issues as tightly to repos...