ros2/rosidl #492: Generate static documentation for interface packages

ahcorde, 2020-05-20:
I have been working on a package that I have called ros2_interfaces_documentation. The aim of this package is to generate the static documentation related with interfaces: messages, services and actions. The same kind of static site that we had for ROS, for example: [sensor_msgs/Imu](http://docs.ros.org/melodic/api/sensor_msgs/html/msg/Imu.html). This functionality is mainly only of use to those maintaining documentation Web sites.

This packages is needed to to reach Quality Level 2 in the interface packages.

This package uses `rosidl_parser` and `rosidl_runtime_py` to get all the available interfaces in the ROS 2 workspace and it will generate a folder for each packages with the following structure:

```
sensor_msgs/
    msg/
         msg1.msg
         msg2.msg
         ...
    action/
        action1.action
        action2.action
        ....
    srv/
        srv1.srv
        srv2.srv
    ....
    msg-index.html
    srv-index.html
    action-index.html
std_msgs/
  ...
```

Then these folders can be added to the [docs.ros2.org repository](https://github.com/ros2/docs.ros2.org).

Where I should place this new package? rosidl? [https://github.com/ros2/rosidl_runtime_py](rosidl_runtime_py) ? or a new repo (For example: ros2/ros2_docuementation_generator) ?

@dirk-thomas and @tfoote what do you think?

dirk-thomas, 2020-05-20:
In a separate repository. After creating it please make sure that all content is added through pull requests (rather then already existing commits).

ahcorde, 2020-05-20:
@dirk-thomas  can you create this repository? I don't have permissions.

> fter creating it please make sure that all content is added through pull requests (rather then already existing commits).

Sure :factory_worker:

dirk-thomas, 2020-05-20:
> can you create this repository?

What should the name be? `ros2_documentation_generator` sounds too generic if it only documents message / services / actions.

ahcorde, 2020-05-26:
Initial PR https://github.com/ros2/ros2_generate_interface_docs/pull/1

Closing this

closed 2020-05-26 by ahcorde