ros2/rosidl #497: TypeError in template: msg__functions.c.em

ShivamPR21, 2020-05-31:
## Build Error

I was trying to build rosidl from source using ```colcon build``` and got an error for function ```idl_structure_type_to_c_include_prefix()``` saying that it takes 1 positional argument but got 2.

From the definition, it takes 2 arguments.
```
def idl_structure_type_to_c_include_prefix(namespaced_type, subdirectory=None):
    parts = [
        convert_camel_case_to_lower_case_underscore(x)
        for x in (namespaced_type.namespaced_name())]
    if subdirectory is not None:
        parts[-1:-1] = [subdirectory]
    include_prefix = '/'.join(parts)
    # Strip service or action suffixes
    if include_prefix.endswith('__request'):
        include_prefix = include_prefix[:-9]
    elif include_prefix.endswith('__response'):
        include_prefix = include_prefix[:-10]
    if include_prefix.endswith('__goal'):
        include_prefix = include_prefix[:-6]
    elif include_prefix.endswith('__result'):
        include_prefix = include_prefix[:-8]
    elif include_prefix.endswith('__feedback'):
        include_prefix = include_prefix[:-10]
    return include_prefix
``` 
And call to function also seems to be fine 
```
include_prefix = idl_structure_type_to_c_include_prefix(
          type_, 'detail')
```

but still getting the error, here is my terminal output.

``` 
--- stderr: rosidl_generator_c                                             
TypeError in template '/home/shivam/sandbox/ridl/src/rosidl/rosidl_generator_c/resource/msg__functions.c.em': idl_structure_type_to_c_include_prefix() takes 1 positional argument but 2 were given
TypeError when expanding 'idl__functions.c.em' into '/home/shivam/sandbox/ridl/build/rosidl_generator_c/rosidl_generator_c/rosidl_generator_c/msg/arrays__functions.c': idl_structure_type_to_c_include_prefix() takes 1 positional argument but 2 were given
Error processing idl file: /home/shivam/sandbox/ridl/build/rosidl_generator_c/rosidl_adapter/rosidl_generator_c/msg/Arrays.idl
Traceback (most recent call last):
  File "/home/shivam/sandbox/ridl/src/rosidl/rosidl_generator_c/bin/rosidl_generator_c", line 40, in <module>
    sys.exit(main())
  File "/home/shivam/sandbox/ridl/src/rosidl/rosidl_generator_c/bin/rosidl_generator_c", line 35, in main
    args.generator_arguments_file,
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_generator_c/__init__.py", line 37, in generate_c
    generate_files(generator_arguments_file, mapping)
  File "/home/shivam/sandbox/ridl/install/rosidl_cmake/lib/python3.6/site-packages/rosidl_cmake/__init__.py", line 94, in generate_files
    raise(e)
  File "/home/shivam/sandbox/ridl/install/rosidl_cmake/lib/python3.6/site-packages/rosidl_cmake/__init__.py", line 89, in generate_files
    post_process_callback=post_process_callback)
  File "/home/shivam/sandbox/ridl/install/rosidl_cmake/lib/python3.6/site-packages/rosidl_cmake/__init__.py", line 147, in expand_template
    interpreter.string(template_content, template_path, locals=data)
  File "/home/shivam/anaconda3/envs/RoboTech4newIndia/lib/python3.6/site-packages/em.py", line 2391, in string
    self.safe(scanner, True, locals)
  File "/home/shivam/anaconda3/envs/RoboTech4newIndia/lib/python3.6/site-packages/em.py", line 2401, in safe
    self.parse(scanner, locals)
  File "/home/shivam/anaconda3/envs/RoboTech4newIndia/lib/python3.6/site-packages/em.py", line 2421, in parse
    token.run(self, locals)
  File "/home/shivam/anaconda3/envs/RoboTech4newIndia/lib/python3.6/site-packages/em.py", line 1547, in run
    self.subrun(info[0][1], interpreter, locals)
  File "/home/shivam/anaconda3/envs/RoboTech4newIndia/lib/python3.6/site-packages/em.py", line 1631, in subrun
    token.run(interpreter, locals)
  File "/home/shivam/anaconda3/envs/RoboTech4newIndia/lib/python3.6/site-packages/em.py", line 1425, in run
    interpreter.execute(self.code, locals)
  File "/home/shivam/anaconda3/envs/RoboTech4newIndia/lib/python3.6/site-packages/em.py", line 2595, in execute
    _exec(statements, self.globals, locals)
  File "<string>", line 5, in <module>
  File "/home/shivam/sandbox/ridl/install/rosidl_cmake/lib/python3.6/site-packages/rosidl_cmake/__init__.py", line 196, in _expand_template
    interpreter.string(content, str(template_path), kwargs)
  File "/home/shivam/anaconda3/envs/RoboTech4newIndia/lib/python3.6/site-packages/em.py", line 2391, in string
    self.safe(scanner, True, locals)
  File "/home/shivam/anaconda3/envs/RoboTech4newIndia/lib/python3.6/site-packages/em.py", line 2401, in safe
    self.parse(scanner, locals)
  File "/home/shivam/anaconda3/envs/RoboTech4newIndia/lib/python3.6/site-packages/em.py", line 2421, in parse
    token.run(self, locals)
  File "/home/shivam/anaconda3/envs/RoboTech4newIndia/lib/python3.6/site-packages/em.py", line 1425, in run
    interpreter.execute(self.code, locals)
  File "/home/shivam/anaconda3/envs/RoboTech4newIndia/lib/python3.6/site-packages/em.py", line 2595, in execute
    _exec(statements, self.globals, locals)
  File "<string>", line 22, in <module>
TypeError: idl_structure_type_to_c_include_prefix() takes 1 positional argument but 2 were given
Exception ignored in: <bound method Interpreter.__del__ of <empy pseudomodule/interpreter at 0x7ff9d6285400>>
Traceback (most recent call last):
  File "/home/shivam/anaconda3/envs/RoboTech4newIndia/lib/python3.6/site-packages/em.py", line 2094, in __del__
  File "/home/shivam/anaconda3/envs/RoboTech4newIndia/lib/python3.6/site-packages/em.py", line 2157, in shutdown
  File "/home/shivam/anaconda3/envs/RoboTech4newIndia/lib/python3.6/site-packages/em.py", line 2646, in finalize
  File "/home/shivam/anaconda3/envs/RoboTech4newIndia/lib/python3.6/site-packages/em.py", line 2200, in push
AttributeError: '_io.TextIOWrapper' object has no attribute 'push'
make[2]: *** [rosidl_generator_c/rosidl_generator_c/msg/arrays.h] Error 1
make[2]: *** Deleting file 'rosidl_generator_c/rosidl_generator_c/msg/arrays.h'
make[1]: *** [CMakeFiles/rosidl_generator_c_interfaces__rosidl_generator_c.dir/all] Error 2
make: *** [all] Error 2
---

```

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  -master
- DDS implementation:
  - Fast-RTPS

clalancette, 2020-06-01:
I suspect that either:

1.  You are mixing some older ROS 2 code (like Eloquent) with newer ROS 2 code (master), or
1.  Something is interacting weirdly in your anaconda environment.

Can you please run `vcs export src --exact-with-tags` on your workspace and post the result here?

ShivamPR21, 2020-06-01:
@clalancette yes, you are right I have ros dashing installed as prebuild package but I have tried to build all of its dependencies from source on the branch master.

Here is my ```vcs export ``` output for all the workspaces that are sourced during the build of rosidl:-

```
repositories:
  rosidl:
    type: git
    url: https://github.com/ros2/rosidl.git
    version: 8518743dd3ff673a91fed1e66e15e58f0e051b9c
```

```
repositories:
  launch:
    type: git
    url: https://github.com/ros2/launch.git
    version: 0.10.2
  osrf_testing_tools_cpp:
    type: git
    url: https://github.com/osrf/osrf_testing_tools_cpp.git
    version: 1.3.2
  rcutils:
    type: git
    url: https://github.com/ros2/rcutils.git
    version: 1.0.0
```
```
repositories:
  test_interface_files:
    type: git
    url: https://github.com/ros2/test_interface_files.git
    version: 0.8.0
```

clalancette, 2020-06-01:
Unfortunately, that is not going to work.  You can't use an older ROS 2 distribution (Dashing) with the newest state of the code; too many things have changed.  If you'd like to rebuild rosidl, you'll need to start from the version released in dashing, i.e. 0.7.9.

I'm going to close this out for now, but feel free to keep commenting.

closed 2020-06-01 by clalancette