ros2/rosidl #501: Resulting IDL conflicts with OMG IDL spec

adamsj-oci, 2020-06-22:
The resulting IDL from the `UUID.msg` seems to conflict with the [IDL spec Section 7.5.3](https://www.omg.org/spec/IDL/4.2/PDF). 

I realize there are some expectations from a ROS developer creating MSG files not to have to concern themselves with IDL specs. This is probably the wrong level to deal with this type of conflict. However, I feel it's worth calling this issue out at this level to start the conversation on how to remedy such conflicts since the layers between aren't handling it.

adamsj-oci, 2020-06-22:
Makes sense. I suspect some vendors are not properly flagging this as a warning or error. However, with broader adoption means that types could be defined with this issue and then it becomes harder to remove. Still I agree that it's a lower priority.

mitza-oci, 2020-06-22:
This isn't really a "vendor" issue though, is it?  If rosidl is responsible for the `.msg` -> `.idl` transformation, then the resulting `.idl` should be valid according to the OMG (and ISO) IDL specification so it can be used with any tool that consumes IDL even if it's not specific to ROS2 (modeling tools, for example).

dirk-thomas, 2020-06-22:
Indeed, but the urgency is determined by the fact that it is / is not affecting used tools. Since afaik all RMW impl. work just fine I consider it a lower priority. If you are aware of a tool which has a problem with it that would probably bump the urgency.

mitza-oci, 2020-06-22:
Right, I think we're not blocked by it, but only by using a switch on the IDL compiler that allows nonconforming input (@adamsj-oci can confirm).  

The other point I was trying to make before is that one advantage of using IDL is that it can be processed by tools outside of the ROS2 system.  So someone could be having the problem and we wouldn't know.  Perhaps they'd file the issue, or perhaps they would just think that ROS2's IDL is a different "dialect" of IDL.

dirk-thomas, 2020-06-24:
> only by using a switch on the IDL compiler that allows nonconforming input

Can you please state which concrete IDL compiler you are referring to.

adamsj-oci, 2020-06-26:
The `tao_idl` compiler shows this behavior.

adamsj-oci, 2020-06-29:
We actually use two different IDL compilers (`tao_idl` and `opendds_idl`) in different phases. The both [indicate the problem](https://dev.azure.com/rmw-opendds/rmw_build/_build/results?buildId=702&view=logs&j=0da5d1d9-276d-5173-c4c4-9d4d4ed14fdb&t=47d55512-6160-5b27-084a-e3e9b13d4d4b&l=40).

dirk-thomas, 2020-07-23:
I would suggest to add a warning about this. It would be great if someone could contribute a pull request for this.