ros2/rosidl #510: Request to add explicit linker language to all rosidl-related packages

atyshka, 2020-07-29:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - Latest master
- DDS implementation:
  - All
- Client library (if applicable):
  - General

----
## Feature request

#### Feature description
When exporting targets from rosidl generators using ament_export_targets, I am running into issues with language ambiguity. I am currently working on a Swift generator, which exports generated swift files. This is apparently creating issues with dependent packages choosing the wrong linker language. For example, std_msgs is exporting generated swift messages along with other (C, c++) generated messages. However, when the python generator declares a dependency on std_msgs for its core C code, CMake sees that swift files were exported from std_msgs and incorrectly chooses Swift to compile the C code for python. 

#### Implementation considerations

Unless a better solution is proposed, I'm thinking the best option is to manually declare the linker language on all targets that link to generated files, using set_target_properties(target_name PROPERTIES LINKER_LANGUAGE C) (or CXX)

dirk-thomas, 2020-07-31:
> I'm thinking the best option is to manually declare the linker language on all targets that link to generated files, using set_target_properties(target_name PROPERTIES LINKER_LANGUAGE C) (or CXX)

Sounds reasonable to me. Please consider contributing pull requests to update the existing rosidl generators with this property.

atyshka, 2020-07-31:
I also noticed that the typesupports link against these and would also need the fix. Is there a reason why the typesupports are dependent on the generated files? I thought since they were lower level than the generators they wouldn’t be dependent on them

dirk-thomas, 2020-07-31:
The typesupport generators map between the ROS data type in a specific language to the RMW vendor specific data type. So naturally they are one level *above* the message generators.

atyshka, 2020-08-01:
Okay, thanks for the clarification. Unfortunately the problem seems too extensive to try and fix this way. It's not only generators/typesupport, any package which uses the generated messages must also declare the linker language explicitly to avoid the issue. That simply isn't realistic. Have you ever worked with CMake languages other than C/C++? I am still stumped as to why this is such an issue with Swift. Is it rather standard practice in CMake to link to targets you don't plan to use? Is it fair to expect CMake to automatically select the correct language when multiple languages are linked to a target, or does this fall into the realm of undefined behavior? I filed a bug report on CMake but it seems help there is minimal given that reproducing it currently requires the whole ros setup. Working on reproducing the issue in vanilla CMake, hopefully it can be solved on that end (although best case scenario that would be fixed in CMake 3.19, which won't be default on Ubuntu).

atyshka, 2020-08-01:
@dirk-thomas I have found a potential alternative. Not sure if you're familiar with it, but in multi-language targets CMake uses CMAKE_<lang>_LINKER_PREFERENCE variables, whichever language is the highest is chose. Digging through CMake source, the default for these variables is 10 for C, 20, for C++ , and 50 for Swift. This is why CMake is choosing swift automatically for any language which links against a swift target. I know that within a package I can easily set the CMAKE_SWIFT_LINKER_PREFERENCE to something low like 5, but that doesn't propagate to packages that depend on it. I.e. from std_msgs I would like to tell all packages depending on this to set CMAKE_SWIFT_LINKER_PREFERENCE to 5.  Is there any way to export variables like this to dependent packages?

atyshka, 2020-08-01:
Update: I got almost everything working by using CONFIG_EXTRAS. This exports a cmake file, which in turn exports another cmake file that correctly sets the language preferences. This works for targets in all dependencies... except for those added for ament_add_gtest. Debugging, with the example of test_msgs, I find that at line 38, `  ament_add_gtest(test_action_typesupport_c_builds ...`, if I print out the LINKER_PREFERENCE variables for C++ and Swift, they end up being 20 and 0, respectively. C++, being higher, should therefore be the chosen language. But for some reason, test are behaving differently.

dirk-thomas, 2020-08-03:
> I got almost everything working by using CONFIG_EXTRAS.

Where did you add the extra CMake logic? It should be in the swift generator package imo.

atyshka, 2020-08-03:
Yes it’s in the Swift generator. So for example, std msgs depends on Swift generator, so it gets config extra A. Config extra A then directs std_msgs to export config extra B, which sets the linker language preference for all packages depending on std_msgs.

atyshka, 2020-08-03:
Why wouldn’t it work for the tests though? Do those somehow have different variable scope?

dirk-thomas, 2020-08-03:
> Why wouldn’t it work for the tests though?

I don't have an explanation at hand. Maybe try to identify what is different in for the targets cases where it fails.

atyshka, 2020-08-03:
Alright, thanks for the assistance thus far, I’ll try to do some further investigation with the tests. Since this issue isn’t relevant anymore I’ll close it.

closed 2020-08-03 by atyshka