ros2/rosidl #520: Dependecy order for custom typesupport

pablogs9, 2020-09-01:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04 (ros:foxy docker)
- Installation type:
  - binaries
- Version or commit hash:
  - foxy

#### Steps to reproduce issue

```
docker run -it --rm ros:foxy
mkdir -p ws/src; cd ws;

source /opt/ros/foxy/setup.bash 

git clone -b foxy https://github.com/micro-ROS/rosidl_typesupport_microxrcedds src/rosidl_typesupport_microxrcedds
git clone -b foxy https://github.com/ros2/common_interfaces src/common_interfaces
git clone -b foxy https://github.com/eProsima/Micro-CDR src/Micro-CDR
git clone -b foxy https://github.com/eProsima/Micro-XRCE-DDS-Client src/Micro-XRCE-DDS-Client

colcon build
```

#### Expected behavior

The typesupport in `src/rosidl_typesupport_microxrcedds` has `<member_of_group>rosidl_typesupport_c_packages</member_of_group>` in its `package.xml` so we expect that it is recognized as a typesupport provider and build and register before all messages contained in the workspace. 

So, in the end, we should have libraries for this typesupport for each message.

#### Actual behavior

`rosidl_typesupport_microxrcedds_c` is not built and register before the interfaces, so libraries for this typesupport are not built:

```bash
# colcon build                                                                                                                                                                                    
Starting >>> std_msgs                                                                                                                                                                                              
Starting >>> microcdr                                                                                                                                                                                              
Starting >>> std_srvs                                                                                                                                                                                              
Finished <<< microcdr [1.16s]                                                                                                                                                                                      
Starting >>> rosidl_typesupport_microxrcedds_c                                                                                                                                                                     
Starting >>> microxrcedds_client                                                                                                                                                                                   
Finished <<< rosidl_typesupport_microxrcedds_c [1.49s]                                                                                                                                                             
Starting >>> rosidl_typesupport_microxrcedds_cpp                                                                                                                                                                   
Finished <<< microxrcedds_client [2.44s]                                                                                                                                                                           
Finished <<< rosidl_typesupport_microxrcedds_cpp [1.62s]                                                                                                                                                           
Starting >>> rosidl_typesupport_microxrcedds_test_msg                                                                                                                                                              
Finished <<< std_srvs [7.85s]                                                                                                                                                                                      
Finished <<< rosidl_typesupport_microxrcedds_test_msg [13.7s]                                                                                                                                                      
Starting >>> rosidl_typesupport_microxrcedds_c_tests                                                                                                                                                               
Starting >>> rosidl_typesupport_microxrcedds_cpp_tests                                                                                                                                                             
Finished <<< rosidl_typesupport_microxrcedds_c_tests [7.90s]                                                                                                                                                       
Finished <<< rosidl_typesupport_microxrcedds_cpp_tests [7.99s]                                                                                                                                                     
Finished <<< std_msgs [29.1s]                                                                                                                                                                                      
Starting >>> geometry_msgs                                                                                                                                                                                         
Starting >>> actionlib_msgs                                                                                                                                                                                        
Finished <<< actionlib_msgs [9.29s]                                                                                                                                                                                
Finished <<< geometry_msgs [24.4s]
Starting >>> sensor_msgs
Starting >>> diagnostic_msgs
Starting >>> nav_msgs
Starting >>> shape_msgs
Starting >>> trajectory_msgs
Starting >>> visualization_msgs
Finished <<< shape_msgs [23.9s]
Finished <<< trajectory_msgs [24.6s]
Finished <<< diagnostic_msgs [26.3s]
Finished <<< nav_msgs [35.0s]
Finished <<< visualization_msgs [50.3s]
Finished <<< sensor_msgs [1min 3s]
Starting >>> stereo_msgs
Finished <<< stereo_msgs [5.13s]
Starting >>> common_interfaces
Finished <<< common_interfaces [0.37s]

# ls install/std_msgs/lib/
libstd_msgs__python.so              libstd_msgs__rosidl_typesupport_c.so    libstd_msgs__rosidl_typesupport_fastrtps_c.so    libstd_msgs__rosidl_typesupport_introspection_c.so    python3.8
libstd_msgs__rosidl_generator_c.so  libstd_msgs__rosidl_typesupport_cpp.so  libstd_msgs__rosidl_typesupport_fastrtps_cpp.so  libstd_msgs__rosidl_typesupport_introspection_cpp.so

```

If we build first the typesupport and the rest of the workspace it works as expected:

```bash
# rm -rf build/ install/ log/

# colcon build --packages-up-to rosidl_typesupport_microxrcedds_c
[...]

# source install/local_setup.bash
# colcon build
[...]

# ls install/std_msgs/lib/
libstd_msgs__python.so                libstd_msgs__rosidl_typesupport_cpp.so           libstd_msgs__rosidl_typesupport_introspection_c.so    python3.8
libstd_msgs__rosidl_generator_c.so    libstd_msgs__rosidl_typesupport_fastrtps_c.so    libstd_msgs__rosidl_typesupport_introspection_cpp.so
libstd_msgs__rosidl_typesupport_c.so  libstd_msgs__rosidl_typesupport_fastrtps_cpp.so  libstd_msgs__rosidl_typesupport_microxrcedds_c.a
```

Notice `libstd_msgs__rosidl_typesupport_microxrcedds_c.a`

#### Additional information

dirk-thomas, 2020-09-10:
For a partial workspace like this I don't think you can expect this to work as described. You basically leave out the middle of the dependency graph.

To make this work you need to also build the other packages involved in the rosidl generation:
* https://github.com/ros2/rosidl.git
* https://github.com/rosidl/rosidl_typesupport.git
* https://github.com/ros2/rosidl_defaults.git

With these additional repositories cloned into your workspace you should see that the message packages are now having a dependency on the custom typesupport packages: e.g. by running `colcon graph`.

pablogs9, 2020-09-11:
Does this mean that if I want to use a middleware not included in the ROS 2 Foxy installation I need to duplicate already installed packages in my workspace?

dirk-thomas, 2020-09-11:
It doesn't mean that you have to duplicate all packages already installed. But for the rosidl packages to pick up additional generators for other languages or typesupport for additional RMW some must be present if you want to rely on the build process to figure out the order for you.

As an alternative you could also build just the custom typesupport package in a separate workspace A before rebuilding all message packages in an overlay workspace B (as you have done in micro-ROS/micro_ros_setup#186). That way the order is explicit and you avoid to rebuild the other rosidl packages.

dirk-thomas, 2020-10-01:
Assuming my previous comments have answered your questions I will go ahead and close the ticket. Please feel free to continue commenting on it.

closed 2020-10-01 by dirk-thomas