ros2/rosidl #544: Foxy package inconsistency

zultron, 2020-11-04:
The `ros-foxy-rosidl-runtime-cpp` package currently shipping in Foxy is v. `1.0.1-1focal.20200624.054926`.  The `rosidl_generator_traits::name()` template function was added in #512 and later tagged in `1.1.0`, so that function isn't declared in `/opt/ros/foxy/include/rosidl_runtime_cpp/traits.hpp`.

However, the `ros-foxy-test-msgs` package version `1.0.0-1focal.20201007.204652` ships a `fibonacci` action with headers generated against some post-#512 revision, so it uses the `name()` template even though it's not provided in the `ros-foxy-rosidl-runtime-cpp` package, and the compiler errors out; see below.

## Bug report

- Operating System:
  - Ubuntu Focal
- Installation type:
  - Binary packages
- Version or commit hash:
  - Package ver. `1.0.1-1focal.20200624.054926`
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
```
docker run -it osrf/ros:foxy-desktop bash -xec '
    apt-get update
    apt-get install ros-foxy-test-msgs
    echo "#include \"test_msgs/action/fibonacci.hpp\"" > foo.cpp
    g++ -I/opt/ros/foxy/include -o foo.o -c foo.cpp
'
```

#### Expected behavior

Program should compile successfully

#### Actual behavior

Compile fails:
```
In file included from /opt/ros/foxy/include/test_msgs/action/fibonacci.hpp:9,
                 from foo.cpp:1:
/opt/ros/foxy/include/test_msgs/action/detail/fibonacci__traits.hpp:23:25: error: expected initializer before ‘<’ token
   23 | inline const char * name<test_msgs::action::Fibonacci_Goal>()
      |                         ^
/opt/ros/foxy/include/test_msgs/action/detail/fibonacci__traits.hpp:52:25: error: expected initializer before ‘<’ token
   52 | inline const char * name<test_msgs::action::Fibonacci_Result>()
      |                         ^
[...]
```

#### Additional information

I'm not sure this is really this project's problem, since it really seems related to the build farm, and I will keep looking for the culprit.

However, it may help in case others have problems (e.g. [this answers.ros.org question][1]); a workaround would also be to release `1.1.0` to Foxy.

[1]: https://answers.ros.org/question/361814/colcon-build-failed-to-build-nav2_util/

hidmic, 2020-11-04:
@zultron latest `ros-foxy-rosidl-runtime-cpp` version shipped is `1.1.0-1focal.20201006.193603`. Could it be that you did a partial upgrade or didn't update our `apt` sources? Though I give you that it'd be nice if installing or updating `ros-foxy-test-msgs` would force a `ros-foxy-rosidl-runtime-cpp` update (which is not the case right now).

zultron, 2020-11-05:
You're totally right!  I'm very sorry for wasting your bandwidth.  Problem solved.

closed 2020-11-05 by zultron