ros2/rosidl #545: CMake export a target that is using rosidl_target_interfaces.

MaximilienNaveau, 2020-11-05:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04 / RT_PREEMPT
- Installation type:
  - ROS eloquent binaries
- Version or commit hash:
  - unknown?
- DDS implementation:
  - unknown? 
- Client library (if applicable):
  - rclcpp, **rosidl_default_generators**

#### Steps to reproduce issue

CMakeLists.txt
```
...

find_package(rosidl_default_generators REQUIRED)
rosidl_generate_interfaces(${PROJECT_NAME}_msg_srv "msg/MyMessage.msg")
...
add_library(mylib SHARED my_cpp_file.cpp)
target_include_directories(
  mylib PUBLIC $<BUILD_INTERFACE:${PROJECT_SOURCE_DIR}/include>
               $<INSTALL_INTERFACE:include>)
rosidl_target_interfaces(mylib ${PROJECT_NAME}_msg_srv "rosidl_typesupport_cpp")
...
# install and export
ament_export_interfaces(export_${PROJECT_NAME} HAS_LIBRARY_TARGET)
install(
  TARGETS mylib
  EXPORT export_${PROJECT_NAME}
  LIBRARY DESTINATION lib
  ARCHIVE DESTINATION lib
  RUNTIME DESTINATION lib/${PROJECT_NAME}
  INCLUDES
  DESTINATION include)
```

#### Expected behavior

Successfull build.

#### Actual behavior

```
CMake Error in src/CMakeLists.txt:
  Target "my_package" INTERFACE_INCLUDE_DIRECTORIES property
  contains path:

    "<full_path_to>/workspace_ros2/build/my_package/rosidl_generator_cpp"

  which is prefixed in the build directory.
```

#### Additional information

I do think the problem comes from here: https://github.com/ros2/rosidl/blob/eloquent/rosidl_cmake/cmake/rosidl_target_interfaces.cmake#L52

This:
```
target_include_directories(${target} PUBLIC ${include_directories})
```
should become this:
```
target_include_directories(${target} PUBLIC $<BUILD_INTERFACE:${include_directories}>)
```

Or maybe I am not using thiese tools properly...

sloretz, 2021-12-07:
The function `rosidl_target_interfaces()` is being deprecated in the next release in favor of a new function `rosidl_get_typesupport_target()`. I don't think that will be backported to ROS Eloquent, but it's very easy to copy what it does - adds two underscores between the generation target name and the type support name.

https://github.com/ros2/rosidl/blob/e76ed132fb51d825cc6765059c13664bff0130a1/rosidl_cmake/cmake/rosidl_get_typesupport_target.cmake#L27-L40

A solution might look like this:

```cmake
rosidl_generate_interfaces(${PROJECT_NAME}_msg_srv "msg/MyMessage.msg")
...
add_library(mylib SHARED my_cpp_file.cpp)
...
target_link_libraries(mylib PUBLIC "${PROJECT_NAME}_msg_srv__rosidl_typesupport_cpp")
```

https://github.com/ros2/ros2_documentation/pull/1798

I'll close this since ROS Eloquent is EOL and the problematic function is already deprecated in future ROS releases

closed 2021-12-07 by sloretz

MaximilienNaveau, 2021-12-08:
Thank you very much for the answer!