ros2/rosidl #552: to_yaml and value_to_yaml errors

mkolodziejczyk-piap, 2020-11-12:
Hi,

I have built latest ROS2 from source following instructions from https://index.ros.org/doc/ros2/Installation/Rolling/Linux-Development-Setup/#linux-latest .
Then I've built successfuly navigation2 from source following https://navigation.ros.org/build_instructions/index.html#manually-build-main

So so far I have the following workspaces:
~/ros2_rolling
~/nav2_depend_ws [navigation2 dependencies]
~/navigation2_ws

In navigation2_ws there are nav2_msgs that are built successfully.

Then I tried to build my own workspace with two packages, let's say my_msgs and my_package. my_package is using my_msg and nav2_msgs (just for test)
Unfortunatelly I receive plenty of this kind of errors:
.../nav2_msgs/msg/detail/costmap_filter_info__traits.hpp:41:5: error: ‘value_to_yaml’ was not declared in this scope
or
.../my_msgs/msg/detail/some_msg__traits.hpp:32:45: error: no matching function for call to ‘to_yaml(const _header_type&, std::ostream&, size_t)’

I searched that those yaml functions are quite new to rosidl, but I've got no idea how compiling messages works to be able to solve this problem.

I'm surprised that nav2_msgs works fine from navigation2_ws, but either my own msgs or nav2_msgs don't work from my workspace. Seems like I'm missing something not in message definition but in message usage in my package

I didn't have these problem on foxy

Regards,

clalancette, 2020-11-12:
I don't know if you've seen our contributing guidelines, but please ask these kind of questions on https://answers.ros.org.  The reason is so that we have questions and answers all in the same, searchable place.

For now I'm going to close this out, but feel free to reopen if you find a problem that needs to be fixed.

closed 2020-11-12 by clalancette

ipa-hsd, 2020-11-17:
I am having the same problem. 
```
In file included from /home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/transform__traits.hpp:17,
                 from /home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/transform_stamped__traits.hpp:19,
                 from /home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/transform_stamped.hpp:9,
                 from /home/user/bridge_ws/src/ros2/geometry2/tf2/include/tf2/buffer_core.h:47,
                 from /home/user/bridge_ws/src/ros2/geometry2/tf2/src/buffer_core.cpp:39:
/home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/vector3__traits.hpp: In function ‘void rosidl_generator_traits::to_yaml(const Vector3&, std::ostream&, size_t)’:
/home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/vector3__traits.hpp:28:5: error: ‘value_to_yaml’ was not declared in this scope
   28 |     value_to_yaml(msg.x, out);
      |     ^~~~~~~~~~~~~
/home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/vector3__traits.hpp:38:5: error: ‘value_to_yaml’ was not declared in this scope
   38 |     value_to_yaml(msg.y, out);
      |     ^~~~~~~~~~~~~
/home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/vector3__traits.hpp:48:5: error: ‘value_to_yaml’ was not declared in this scope
   48 |     value_to_yaml(msg.z, out);
      |     ^~~~~~~~~~~~~
In file included from /home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/transform__traits.hpp:19,
                 from /home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/transform_stamped__traits.hpp:19,
                 from /home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/transform_stamped.hpp:9,
                 from /home/user/bridge_ws/src/ros2/geometry2/tf2/include/tf2/buffer_core.h:47,
                 from /home/user/bridge_ws/src/ros2/geometry2/tf2/src/buffer_core.cpp:39:
/home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/quaternion__traits.hpp: In function ‘void rosidl_generator_traits::to_yaml(const Quaternion&, std::ostream&, size_t)’:
/home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/quaternion__traits.hpp:28:5: error: ‘value_to_yaml’ was not declared in this scope
   28 |     value_to_yaml(msg.x, out);
      |     ^~~~~~~~~~~~~
/home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/quaternion__traits.hpp:38:5: error: ‘value_to_yaml’ was not declared in this scope
   38 |     value_to_yaml(msg.y, out);
      |     ^~~~~~~~~~~~~
/home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/quaternion__traits.hpp:48:5: error: ‘value_to_yaml’ was not declared in this scope
   48 |     value_to_yaml(msg.z, out);
      |     ^~~~~~~~~~~~~
/home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/quaternion__traits.hpp:58:5: error: ‘value_to_yaml’ was not declared in this scope
   58 |     value_to_yaml(msg.w, out);
      |     ^~~~~~~~~~~~~
In file included from /home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/transform_stamped.hpp:9,
                 from /home/user/bridge_ws/src/ros2/geometry2/tf2/include/tf2/buffer_core.h:47,
                 from /home/user/bridge_ws/src/ros2/geometry2/tf2/src/buffer_core.cpp:39:
/home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/transform_stamped__traits.hpp: In function ‘void rosidl_generator_traits::to_yaml(const TransformStamped&, std::ostream&, size_t)’:
/home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/transform_stamped__traits.hpp:34:45: error: no matching function for call to ‘to_yaml(const _header_type&, std::ostream&, size_t)’
   34 |     to_yaml(msg.header, out, indentation + 2);
      |                                             ^
In file included from /home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/transform__traits.hpp:17,
                 from /home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/transform_stamped__traits.hpp:19,
                 from /home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/transform_stamped.hpp:9,
                 from /home/user/bridge_ws/src/ros2/geometry2/tf2/include/tf2/buffer_core.h:47,
                 from /home/user/bridge_ws/src/ros2/geometry2/tf2/src/buffer_core.cpp:39:
/home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/vector3__traits.hpp:18:13: note: candidate: ‘void rosidl_generator_traits::to_yaml(const Vector3&, std::ostream&, size_t)’
   18 | inline void to_yaml(
      |             ^~~~~~~
/home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/vector3__traits.hpp:19:39: note:   no known conversion for argument 1 from ‘const _header_type’ {aka ‘const std_msgs::msg::Header_<std::allocator<void> >’} to ‘const Vector3&’ {aka ‘const geometry_msgs::msg::Vector3_<std::allocator<void> >&’}
   19 |   const geometry_msgs::msg::Vector3 & msg,
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/vector3__traits.hpp:53:20: note: candidate: ‘std::string rosidl_generator_traits::to_yaml(const Vector3&)’
   53 | inline std::string to_yaml(const geometry_msgs::msg::Vector3 & msg)
      |                    ^~~~~~~
[... cut ...]
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/home/user/bridge_ws/install/geometry_msgs/include/geometry_msgs/msg/detail/transform_stamped__traits.hpp:43:5: error: ‘value_to_yaml’ was not declared in this scope
   43 |     value_to_yaml(msg.child_frame_id, out);
      |     ^~~~~~~~~~~~~
make[2]: *** [CMakeFiles/tf2.dir/build.make:76: CMakeFiles/tf2.dir/src/buffer_core.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:163: CMakeFiles/tf2.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

mkolodziejczyk-piap, 2020-11-17:
@clalancette, as we see this is not only my problem.
IMO asking this kind of question on ros answers is pointless:
1. I bet 99% of ROS users doesn't understand the details of rosidl. And in general, "normal" users usually use stable releases. I think the developers of rosidl are the only ones that may help. And if this is really a bug it's better for them to know about it sooner than later
2. It's kind of new feature on "dev" branch. The chances it will have any informative value in the future are low

I've tried lot's of tests and to me it's either a bug or some conflict with my foxy distribution (= probably also a bug because I think I'm doing sourcing right)

Regards,

ipa-hsd, 2020-11-17:
@mkolodziejczyk-piap what worked for me is to use the [lastSuccessfulBuild](https://ci.ros2.org/view/packaging/job/packaging_linux/lastSuccessfulBuild/artifact/ws/ros2-package-linux-x86_64.tar.bz2)

```
$ wget -nv https://ci.ros2.org/view/packaging/job/packaging_linux/lastSuccessfulBuild/artifact/ws/ros2-package-linux-x86_64.tar.bz2
$ tar -xjf ros2-package-linux-x86_64.tar.bz2
```
then you can source the `local_setup.bash` inside this folder.

jediofgever, 2020-12-03:
I am almost sure that this is caused by conflicting 2 or more different ros2 distros, after I remove the deb installed ros2 distro , I no more see this scary error.

ipa-hsd, 2020-12-04:
I am not sure what gets pushed into the master branch, but I have encountered build errors in the past as well when I pull in the master branch. So unless I need to work with the source, I tend to use `lastSuccessfulBuild`. But good to know it worked for you.