ros2/rosidl #574: Rbin_*_rosidl_generator_c builds failing: No module named rosidl_adapter

jacobperron, 2021-02-26:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - 2.0.3

Builds have been failing since Feb 24th: https://build.ros2.org/view/Rbin_uF64/job/Rbin_uF64__rosidl_generator_c__ubuntu_focal_amd64__binary/38/

https://build.ros2.org/view/Rbin_ufv8_uFv8/job/Rbin_ufv8_uFv8__rosidl_generator_c__ubuntu_focal_arm64__binary/30/

```
23:03:45 CMake Error at /opt/ros/rolling/share/rosidl_adapter/cmake/rosidl_adapt_interfaces.cmake:60 (message):
23:03:45   execute_process(/usr/bin/python3 -m rosidl_adapter --package-name
23:03:45   rosidl_generator_c --arguments-file
23:03:45   /tmp/binarydeb/ros-rolling-rosidl-generator-c-2.0.3/obj-x86_64-linux-gnu/rosidl_adapter__arguments__rosidl_generator_c_interfaces.json
23:03:45   --output-dir
23:03:45   /tmp/binarydeb/ros-rolling-rosidl-generator-c-2.0.3/obj-x86_64-linux-gnu/rosidl_adapter/rosidl_generator_c
23:03:45   --output-file
23:03:45   /tmp/binarydeb/ros-rolling-rosidl-generator-c-2.0.3/obj-x86_64-linux-gnu/rosidl_adapter/rosidl_generator_c_interfaces.idls)
23:03:45   returned error code 1:
23:03:45 
23:03:45   /usr/bin/python3: No module named rosidl_adapter
23:03:45 
23:03:45 Call Stack (most recent call first):
23:03:45   /opt/ros/rolling/share/rosidl_cmake/cmake/rosidl_g
```

clalancette, 2021-02-26:
This should be fixed by https://github.com/ament/ament_cmake/pull/323

clalancette, 2021-02-26:
And I just merged https://github.com/ros/rosdistro/pull/28424 .  Once that rebuilds, this should in theory be fixed.  I'll leave this open until the rebuild finishes just in case something else needs to be done.

clalancette, 2021-02-26:
Looks like this is still broken: https://build.ros2.org/view/Rbin_uF64/job/Rbin_uF64__rosidl_generator_c__ubuntu_focal_amd64__binary/39/console

hidmic, 2021-02-26:
FTR the actual problem is way below https://build.ros2.org/view/Rbin_uF64/job/Rbin_uF64__ament_cmake_python__ubuntu_focal_amd64__binary/22/console

Could it be that `dh_auto_install` is using yet another mechanism?

jacobperron, 2021-03-01:
Fixed by https://github.com/ament/ament_cmake/pull/324

closed 2021-03-01 by jacobperron