ros2/rosidl #585: build fail: rosidl_generatorc/cpp and rosidl_runtimec/cpp [ROS2 Foxy Focal]

newcanopies, 2021-04-30:
## Bug report

- Operating System:
  - Ubuntu 20.04
  - binary
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

~/ws_ROS2/src/$ `git clone https://github.com/ros2/rosidl.git && git checkout foxy`
~/ws_ROS2/$ `colcon build --symlink-install --packages-select rosidl_adapter rosidl_parser rosidl_cmake rosidl_typesupport_interface rosidl_typesupport_introspection_c rosidl_typesupport_introspection_cpp
`
_Summary: 6 packages finished_ 

~/ws_ROS2/$ `colcon build --symlink-install --packages-select rosidl_generator_c rosidl_generator_cpp rosidl_runtime_c rosidl_runtime_cpp`
Starting >>> rosidl_generator_c
--- stderr: rosidl_generator_c                         
CMake Error at CMakeLists.txt:31 (find_package):
  By not providing "Findtest_interface_files.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "test_interface_files", but CMake did not find one.

  Could not find a package configuration file provided by
  "test_interface_files" with any of the following names:

    test_interface_filesConfig.cmake
    test_interface_files-config.cmake

  Add the installation prefix of "test_interface_files" to CMAKE_PREFIX_PATH
  or set "test_interface_files_DIR" to a directory containing one of the
  above files.  If "test_interface_files" provides a separate development
  package or SDK, be sure it has been installed.

---
Failed   <<< rosidl_generator_c [2.09s, exited with code 1]
                                
Starting >>> rosidl_runtime_c
--- stderr: rosidl_runtime_c                         
CMake Error at CMakeLists.txt:54 (find_package):
  By not providing "Findperformance_test_fixture.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "performance_test_fixture", but CMake did not find one.

  Could not find a package configuration file provided by
  "performance_test_fixture" with any of the following names:

    performance_test_fixtureConfig.cmake
    performance_test_fixture-config.cmake

  Add the installation prefix of "performance_test_fixture" to
  CMAKE_PREFIX_PATH or set "performance_test_fixture_DIR" to a directory
  containing one of the above files.  If "performance_test_fixture" provides
  a separate development package or SDK, be sure it has been installed.

---
Failed   <<< rosidl_runtime_c [1.92s, exited with code 1]

```

#### Expected behavior

full rosidl package should build

#### Actual behavior

these packages fail: 
**rosidl_generator_c 
rosidl_generator_cpp 
rosidl_runtime_c 
rosidl_runtime_cpp**

#### Additional information
$COLCON_ws
![image](https://user-images.githubusercontent.com/65022111/116659757-5e755800-a992-11eb-89da-afde05e2c096.png)

`sudo apt install ros-foxy-rosidl-typesupport-c `

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ros-foxy-rosidl-typesupport-c is already the newest version (1.0.2-1focal.20210415.021333).
ros-foxy-rosidl-typesupport-c set to manually installed.
The following packages were automatically installed and are no longer required:
  python3-xkit ubuntu-drivers-common
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 8 not upgraded.
```

hidmic, 2021-05-03:
IIRC a binary for package A does not bring package A's build nor test dependencies with it. Have you tried to satisfy them using `rosdep`?

newcanopies, 2021-05-07:
Thanks for this insight on build/test deps. However I run ros dep routinely. Something in the above pictured ws_ROS2 tree is conflicting and won't build no matter what. 

`rosidl` finally built in a fresh colcon_ws all by itself. Not clear what could be the source of interference with other core ros2 packages. 

Thank you!

closed 2021-05-07 by newcanopies