ros2/rosidl #602: Failing to find_package interface packages due to missing target rosidl_typesupport_introspection_cpp

jacobperron, 2021-07-15:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binary
- Version or commit hash:
  - Rolling (d1122a2d2cb0c2d048aad6453badc6bbd17800b5)
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

In an isolated environment, in workspace clone `example_interfaces` and create a dummy package that depends on it:
```
mkdir -p repro_ws/src
cd repreo_ws
git clone https://github.com/ros2/example_interfaces.git src/example_interfaces
# install dependencies with rosdep
rosdep install --rosdistro rolling --from-paths src -i -y
# install ros2pkg for convenience
sudo apt install ros-rolling-ros2pkg
# create package 'foo' that depends on example_interfaces
ros2 pkg create --destination-directory src --dependencies example_interfaces -- foo
```

and build the workspace:

```
colcon build --packages-up-to foo
```

#### Expected behavior

Builds without error.

#### Actual behavior

`find_package(example_interfaces REQUIRED)` fails when building `foo`:

```
Starting >>> foo
--- stderr: foo                          
CMake Error at CMakeLists.txt:10 (find_package):
  Found package configuration file:

    /tmp/repro_ws/install/example_interfaces/share/example_interfaces/cmake/example_interfacesConfig.cmake

  but it set example_interfaces_FOUND to FALSE so package
  "example_interfaces" is considered to be NOT FOUND.  Reason given by
  package:

  The following imported targets are referenced, but are missing:
  example_interfaces::example_interfaces__rosidl_typesupport_introspection_cpp
```

jacobperron, 2021-07-15:
@hidmic I know you've been doing some work recently related to rosidl. Any idea what could be causing this?

Originally noticed this when gazebo_ros_pkgs CI started failing for Rolling, e.g. https://build.ros2.org/job/Rpr__gazebo_ros_pkgs__ubuntu_focal_amd64/77

cc/ @adityapande-1995

hidmic, 2021-07-16:
I cannot reproduce from source, so either something wasn't released or there's an issue with debians.

hidmic, 2021-07-16:
Alright, yeah, we are missing https://github.com/ros/rosdistro/pull/30236. Without it, we can and do run into this target export ordering issue. FYI @nuclearsandwich, I know we're in the middle of a sync but that PR has to go in or it'll break all packages downstream interface packages.

jacobperron, 2021-07-16:
The [sync already happened](https://discourse.ros.org/t/new-packages-for-ros-2-rolling-ridley-2021-07-15/21401); we might want to another sync sooner than later to unbreak folks.

jacobperron, 2021-07-20:
With ros/rosdistro#30236 sync'd, this issue is fixed for me!

closed 2021-07-20 by jacobperron