ros2/rosidl #607: Relocate to_yaml() definition to message namespace

hidmic, 2021-08-02:
## Feature request

#### Feature description

Follow-up after #523. When `to_yaml()` function definitions were pushed into the `rosidl_generator_traits` namespace, argument-dependent lookups were blocked. This makes it difficult to use these overloads in templates -- all overloads must be visible to the compiler when it first hits the template definition for them to be picked up.

Relocating them would simplify their use.

hidmic, 2021-08-02:
@sloretz @clalancette @ivanpauno thoughts?

sloretz, 2021-08-02:
I don't understand - got an example code sample that works in plain code but not in a template?

hidmic, 2021-08-02:
@sloretz see this quick and dirty [example](https://coliru.stacked-crooked.com/a/ad3167dd80a9a138) to show the problem with overloads and namespaces.

sloretz, 2021-08-02:
Ah, meaning the problem is templates defined before the message's type traits header is included can't use `to_yaml(...)`? Is this what you mean by moving the definition to the message namespace?
https://coliru.stacked-crooked.com/a/fe75cd8807bd569d

hidmic, 2021-08-02:
That's one option, but it doesn't have to be a method. https://coliru.stacked-crooked.com/a/9067c566be1f3097 would do as well and we wouldn't have to force formatting details into the `struct`.

hidmic, 2021-08-03:
@sloretz thoughts?

sloretz, 2021-08-03:
Seems reasonable to me! It looks like we could have it in both namespaces without too much trouble (https://coliru.stacked-crooked.com/a/d3f6151e3378f8a4), but it seems better to me to have only one way of calling the same thing.

hidmic, 2021-08-03:
Yeah, we can keep it in `rosidl_generator_traits` to help deprecation. I'll send a patch.

ivanpauno, 2021-08-03:
> That's one option, but it doesn't have to be a method. https://coliru.stacked-crooked.com/a/9067c566be1f3097 would do as well and we wouldn't have to force formatting details into the struct.

+1

> Seems reasonable to me! It looks like we could have it in both namespaces without too much trouble (https://coliru.stacked-crooked.com/a/d3f6151e3378f8a4), but it seems better to me to have only one way of calling the same thing.

+1

> Yeah, we can keep it in rosidl_generator_traits to help deprecation. I'll send a patch.

+1

PR #609 (2021-08-04): Relocate to_yaml() under message namespace.
Closes #607. This enables [argument dependent lookup](https://en.cppreference.com/w/cpp/language/adl). `to_yaml()` overloads under the `rosidl_generator_traits` namespace were kept but deprecated. 

CI up to `rosidl_generator_cpp` (to start with):

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]
closed 2021-08-04 by hidmic