ros2/rosidl #629: Build error in Rolling in rosbridge_suite

jtbandes, 2021-11-05:
Our [CI build for Rolling](https://github.com/RobotWebTools/rosbridge_suite/blob/ros2/.github/workflows/ci.yml) recently started failing due to an error in rosidl_adapt_interfaces.cmake. It appears Python 2 is being used instead of Python 3:

https://github.com/RobotWebTools/rosbridge_suite/runs/4121063131?check_suite_focus=true

```
  /usr/bin/bash -c source /opt/ros/rolling/setup.sh && colcon build --event-handlers console_cohesion+ --symlink-install
  Starting >>> rosapi_msgs
  Starting >>> rosbridge_msgs
  --- output: rosbridge_msgs
  -- The C compiler identification is GNU 9.3.0
  -- The CXX compiler identification is GNU 9.3.0
  -- Detecting C compiler ABI info
  -- Detecting C compiler ABI info - done
  -- Check for working C compiler: /usr/bin/cc - skipped
  -- Detecting C compile features
  -- Detecting C compile features - done
  -- Detecting CXX compiler ABI info
  -- Detecting CXX compiler ABI info - done
  -- Check for working CXX compiler: /usr/bin/c++ - skipped
  -- Detecting CXX compile features
  -- Detecting CXX compile features - done
  -- Found ament_cmake_ros: 0.9.2 (/opt/ros/rolling/share/ament_cmake_ros/cmake)
  -- Found Python3: /usr/bin/python3.8 (found version "3.8.10") found components: Interpreter 
  -- Override CMake install command with custom implementation using symlinks instead of copying resources
  -- Found builtin_interfaces: 1.1.0 (/opt/ros/rolling/share/builtin_interfaces/cmake)
  -- Using single rosidl_typesupport_c: rosidl_typesupport_introspection_c
  -- Found rosidl_adapter: 2.5.0 (/opt/ros/rolling/share/rosidl_adapter/cmake)
  -- Using single rosidl_typesupport_cpp: rosidl_typesupport_introspection_cpp
  -- Found rosidl_default_generators: 1.1.1 (/opt/ros/rolling/share/rosidl_default_generators/cmake)
  -- Found PythonInterp: /usr/bin/python (found version "2.7.18") 
  CMake Error at /opt/ros/rolling/share/rosidl_adapter/cmake/rosidl_adapt_interfaces.cmake:60 (message):
    execute_process(/usr/bin/python -m rosidl_adapter --package-name
    rosbridge_msgs --arguments-file
    /home/runner/work/rosbridge_suite/rosbridge_suite/ros_ws/build/rosbridge_msgs/rosidl_adapter__arguments__rosbridge_msgs.json
    --output-dir
    /home/runner/work/rosbridge_suite/rosbridge_suite/ros_ws/build/rosbridge_msgs/rosidl_adapter/rosbridge_msgs
    --output-file
    /home/runner/work/rosbridge_suite/rosbridge_suite/ros_ws/build/rosbridge_msgs/rosidl_adapter/rosbridge_msgs.idls)
    returned error code 1:
  
    Traceback (most recent call last):
  
      File "/usr/lib/python2.7/runpy.py", line 163, in _run_module_as_main
        mod_name, _Error)
      File "/usr/lib/python2.7/runpy.py", line 111, in _get_module_details
        __import__(mod_name)  # Do not catch exceptions initializing package
      File "/opt/ros/rolling/lib/python3.8/site-packages/rosidl_adapter/__init__.py", line 32
        assert False, f"Unsupported interface type '{interface_file.suffix}'"
                                                                            ^
  
    SyntaxError: invalid syntax
```

Any advice on how to fix this would be appreciated!

clalancette, 2021-11-05:
I suspect that we need to do a new release of rosidl, which hasn't been done recently.  I *believe* that will fix this issue, thanks for reporting.  I'll do a new release of it right now.

jtbandes, 2021-12-03:
Seems like the build is working now.

closed 2021-12-03 by jtbandes