ros2/rosidl #634: Galactic: set CMAKE_CXX_STANDARD to 17

oysstu, 2021-11-22:
Since rclcpp targets c++17 in galactic, rosidl should probably do the same.

https://github.com/ros2/rosidl/blob/7d95d643a7260ed37af4c7803422be9a205d9e51/rosidl_typesupport_introspection_cpp/CMakeLists.txt#L7

There's also a few places where C++14 is used in tests
https://github.com/ros2/rosidl/blob/7d95d643a7260ed37af4c7803422be9a205d9e51/rosidl_typesupport_interface/CMakeLists.txt#L26

https://github.com/ros2/rosidl/blob/7d95d643a7260ed37af4c7803422be9a205d9e51/rosidl_runtime_cpp/CMakeLists.txt#L30

https://github.com/ros2/rosidl/blob/7d95d643a7260ed37af4c7803422be9a205d9e51/rosidl_generator_cpp/CMakeLists.txt#L39

https://github.com/ros2/rosidl/blob/7d95d643a7260ed37af4c7803422be9a205d9e51/rosidl_runtime_c/CMakeLists.txt#L64

aprotyas, 2021-11-22:
Do you mind submitting a quick PR to that effect?

oysstu, 2021-11-23:
Done, see #635. Can open a backport PR to Galactic afterwards

closed 2022-01-23 by oysstu