ros2/rosidl #642: nested for index overwrite in PyEM template

ibnHatab, 2022-01-05:
Same index name used in nested loop. Potentially causing problems fex: removing trailing ',' in struct with conditional '@(i < len(msg.fields) - 1 ? ',')@'

This code :
@[  for **i,** field in enumerate(msg.fields)]@
@[      for **i,** line in enumerate(field.annotations['comment'])]

https://github.com/ros2/rosidl/blob/2ec002300a41290ea69092ae883f4c5c02661939/rosidl_adapter/rosidl_adapter/resource/struct.idl.em#L76

clalancette, 2022-01-05:
Good catch.  Would you mind submitting a PR renaming at least one of them?

ibnHatab, 2022-01-06:
Sure, PR submitted : https://github.com/ros2/rosidl/pull/643
pytest passed on rosidl_adapter but I don't have dev env to isolate and recheck everything.

clalancette, 2022-01-27:
This was fixed by #643, so closing.  Thanks for the contribution!

closed 2022-01-27 by clalancette