ros2/rosidl #676: rosidl_generator_c does not produce replicable generated files

pablogs9, 2022-04-21:
## Bug report

**Required Info:**
- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Binaries
- Version or commit hash:
  - main
- DDS implementation:
  - Fast DDS
- Client library (if applicable):
  - N/A

#### Steps to reproduce the issue

Pull and run a ROS 2 Rolling Docker env:

```bash
docker pull ros:rolling
docker run -it --rm ros:rolling
```

Inside the docker prepare a sample environment:

```
apt update
apt upgrade -y
source /opt/ros/rolling/setup.bash
mkdir -p ws/src && cd ws
git clone -b master https://github.com/ros2/common_interfaces src/common_interfaces
cp -r src/common_interfaces/actionlib_msgs/ src
rm -rf src/common_interfaces/
```

Build and check file output multiple times with:

```bash
rm -rf build/ install/ log/ && colcon build --packages-select actionlib_msgs > /dev/null 2>&1 && md5sum install/actionlib_msgs/include/actionlib_msgs/actionlib_msgs/msg/detail/goal_status__struct.h && md5sum install/actionlib_msgs/share/actionlib_msgs/msg/GoalStatus.idl 
```

As seen below, the generated IDL file is always the same but the header files is modified for each build:

```
#1 
f6d011ce1a542d883a3bd23a6664b2ea  install/actionlib_msgs/include/actionlib_msgs/actionlib_msgs/msg/detail/goal_status__struct.h
cd9baf779987f38a9b7cb07d8a952a09  install/actionlib_msgs/share/actionlib_msgs/msg/GoalStatus.idl
#2 
eaf75edc414175000b168954b5d885fe  install/actionlib_msgs/include/actionlib_msgs/actionlib_msgs/msg/detail/goal_status__struct.h
cd9baf779987f38a9b7cb07d8a952a09  install/actionlib_msgs/share/actionlib_msgs/msg/GoalStatus.idl
#3
a2d2b8acfcbdfe937a94373cb2b65f4e  install/actionlib_msgs/include/actionlib_msgs/actionlib_msgs/msg/detail/goal_status__struct.h
cd9baf779987f38a9b7cb07d8a952a09  install/actionlib_msgs/share/actionlib_msgs/msg/GoalStatus.idl
#4
0029b7956254bd3560bebe99c81663cd  install/actionlib_msgs/include/actionlib_msgs/actionlib_msgs/msg/detail/goal_status__struct.h
cd9baf779987f38a9b7cb07d8a952a09  install/actionlib_msgs/share/actionlib_msgs/msg/GoalStatus.idl
#5
c02b65a4261a540d0b8d598245015fbc  install/actionlib_msgs/include/actionlib_msgs/actionlib_msgs/msg/detail/goal_status__struct.h
cd9baf779987f38a9b7cb07d8a952a09  install/actionlib_msgs/share/actionlib_msgs/msg/GoalStatus.idl
```

#### Expected behavior

Generated code is always the same for every iteration.

#### Actual behavior

Generated code randomly modifies spaces and indentation along with runs.

#### Additional information

It seems to be something related to spaces and indentation:

[image]

pablogs9, 2022-04-21:
Proposed solution: https://github.com/ros2/rosidl/pull/677

pablogs9, 2022-04-26:
Friendly ping @sloretz @clalancette

pablogs9, 2022-05-11:
Any update on this?

EduPonz, 2022-06-03:
Friendly ping @sloretz @clalancette , could you guys take a look at this and the proposed solution? Else, could you point us to someone that can so we can move this forward? Thanks!

sloretz, 2022-07-11:
I'm able to reproduce

```
jammy:root> rm -rf build/actionlib_msgs/ install/actionlib_msgs/ && colcon build --packages-select actionlib_msgs >/dev/null && md5sum install/actionlib_msgs/include/actionlib_msgs/actionlib_msgs/msg/detail/goal_status__struct.h
92421f90c85ae49c9765ce3d339144d6  install/actionlib_msgs/include/actionlib_msgs/actionlib_msgs/msg/detail/goal_status__struct.h
jammy:root> rm -rf build/actionlib_msgs/ install/actionlib_msgs/ && colcon build --packages-select actionlib_msgs >/dev/null && md5sum install/actionlib_msgs/include/actionlib_msgs/actionlib_msgs/msg/detail/goal_status__struct.h
b4251ed188e2c4d096be515092383c63  install/actionlib_msgs/include/actionlib_msgs/actionlib_msgs/msg/detail/goal_status__struct.h
jammy:root> rm -rf build/actionlib_msgs/ install/actionlib_msgs/ && colcon build --packages-select actionlib_msgs >/dev/null && md5sum install/actionlib_msgs/include/actionlib_msgs/actionlib_msgs/msg/detail/goal_status__struct.h
ce09f01fd12e6c7ec0e993f96767c1b1  install/actionlib_msgs/include/actionlib_msgs/actionlib_msgs/msg/detail/goal_status__struct.h
jammy:root> rm -rf build/actionlib_msgs/ install/actionlib_msgs/ && colcon build --packages-select actionlib_msgs >/dev/null && md5sum install/actionlib_msgs/include/actionlib_msgs/actionlib_msgs/msg/detail/goal_status__struct.h
b22f317992074bcbf737a8a976d5a2e9  install/actionlib_msgs/include/actionlib_msgs/actionlib_msgs/msg/detail/goal_status__struct.h
jammy:root> rm -rf build/actionlib_msgs/ install/actionlib_msgs/ && colcon build --packages-select actionlib_msgs >/dev/null && md5sum install/actionlib_msgs/include/actionlib_msgs/actionlib_msgs/msg/detail/goal_status__struct.h
4366dcd56bf4dbfc4a3b5cc83c2e13cf  install/actionlib_msgs/include/actionlib_msgs/actionlib_msgs/msg/detail/goal_status__struct.h
jammy:root> rm -rf build/actionlib_msgs/ install/actionlib_msgs/ && colcon build --packages-select actionlib_msgs >/dev/null && md5sum install/actionlib_msgs/include/actionlib_msgs/actionlib_msgs/msg/detail/goal_status__struct.h
ec9724bc3dde398503b5eb2d78d87c1d  install/actionlib_msgs/include/actionlib_msgs/actionlib_msgs/msg/detail/goal_status__struct.h
jammy:root> rm -rf build/actionlib_msgs/ install/actionlib_msgs/ && colcon build --packages-select actionlib_msgs >/dev/null && md5sum install/actionlib_msgs/include/actionlib_msgs/actionlib_msgs/msg/detail/goal_status__struct.h
130c19ed638dc5cfca646b9f868d96ff  install/actionlib_msgs/include/actionlib_msgs/actionlib_msgs/msg/detail/goal_status__struct.h
jammy:root> rm -rf build/actionlib_msgs/ install/actionlib_msgs/ && colcon build --packages-select actionlib_msgs >/dev/null && md5sum install/actionlib_msgs/include/actionlib_msgs/actionlib_msgs/msg/detail/goal_status__struct.h
03cd8d565dc541ec0fae72644ec01a63  install/actionlib_msgs/include/actionlib_msgs/actionlib_msgs/msg/detail/goal_status__struct.h
jammy:root> rm -rf build/actionlib_msgs/ install/actionlib_msgs/ && colcon build --packages-select actionlib_msgs >/dev/null && md5sum install/actionlib_msgs/include/actionlib_msgs/actionlib_msgs/msg/detail/goal_status__struct.h
b0c7d7aa65c00dbbdffbd7effe77c509  install/actionlib_msgs/include/actionlib_msgs/actionlib_msgs/msg/detail/goal_status__struct.h
```

And I also see the differences are varying white space at the start of some comment lines.

```diff
41c41
<   * and has since completed its execution (Terminal State).
---
>   *   and has since completed its execution (Terminal State).
71c71
<   * because the goal was unattainable or invalid (Terminal State).
---
>   *    because the goal was unattainable or invalid (Terminal State).
81c81
<   *    and has not yet completed execution.
---
>   * and has not yet completed execution.
91c91
<   *    the action server has not yet confirmed that the goal is canceled.
---
>   * the action server has not yet confirmed that the goal is canceled.
111c111
<   *    be sent over the wire by an action server.
---
>   * be sent over the wire by an action server.
```

sloretz, 2022-07-11:
Random inclusion of whitespace affects any string literal, not just comments. This sometimes parses as `"   e"`, and sometimes as `"e"`.

```
module MRE {const string foo = "   e";};
```
[image]
[image]

PR #688 (2022-07-12): Always include whitespace in string literals
Alternative to #677 
Fixes #676 

`rosidl_parser` randomly ignores whitespace at the start of string literals. This example matches the string literal as `Token('__ANON_6', '    e')` or `Token('__ANON_6', 'e')` when run multiple times; however, stripping the whitespace is incorrect.

```python3
from rosidl_parser.parser import get_ast_from_idl_string

idl_str ='''module MRE {
const string foo = "    e";
};'''
print(get_ast_from_idl_string(idl_str))
```

I think the cause is a combination of ignoring whitespace

https://github.com/ros2/rosidl/blob/02 [... cut]
closed 2022-07-12 by sloretz