ros2/rosidl #680: undefined symbol: `[message_typename]__copy`

m2-farzan, 2022-05-10:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 20.04
- Installation type:
  -  Tested on Ubuntu binaries as well as local builds
- Version or commit hash:
  -  galactic (binaries build date: 20220430), (local build HEAD:  4ecf4f8 i.e. release 2.2.2)
- DDS implementation:
  -  Fast-RTPS
- Client library (if applicable):
  -  rclpy

#### Steps to reproduce issue

Not a minimal example but easy to test:
1. `apt-get update && apt-get install ros-galactic-nav2-simple-commander`
2. Run:
    ```
    #!/bin/python3
    
    import rclpy
    from nav2_simple_commander.robot_navigator import BasicNavigator
    
    rclpy.init()
    nav = BasicNavigator()
    ```

#### Expected behavior
Exit cleanly

#### Actual behavior
```
1651914735.930865 [0]    python3: selected interface "tap0" is not multicast-capable: disabling multicast
Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/rosidl_generator_py/import_type_support_impl.py", line 46, in import_type_support
    return importlib.import_module(module_name, package=pkg_name)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 657, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 556, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1166, in create_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
ImportError: /opt/ros/galactic/lib/libnav2_msgs__rosidl_generator_c.so: undefined symbol: geometry_msgs__msg__Pose__copy

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "a.py", line 7, in <module>
    nav = BasicNavigator()
  File "/opt/ros/galactic/lib/python3.8/site-packages/nav2_simple_commander/robot_navigator.py", line 60, in __init__
    self.nav_through_poses_client = ActionClient(self,
  File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/action/client.py", line 148, in __init__
    check_for_type_support(action_type)
  File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/type_support.py", line 29, in check_for_type_support
    msg_or_srv_type.__class__.__import_type_support__()
  File "/opt/ros/galactic/lib/python3.8/site-packages/nav2_msgs/action/_navigate_through_poses.py", line 1319, in __import_type_support__
    module = import_type_support('nav2_msgs')
  File "/opt/ros/galactic/lib/python3.8/site-packages/rosidl_generator_py/import_type_support_impl.py", line 48, in import_type_support
    raise UnsupportedTypeSupport(pkg_name)
rosidl_generator_py.import_type_support_impl.UnsupportedTypeSupport: Could not import 'rosidl_typesupport_c' for package 'nav2_msgs'
Exception ignored in: <function ActionClient.__del__ at 0x7fce4eb1fdc0>
Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/action/client.py", line 597, in __del__
    self.destroy()
  File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/action/client.py", line 588, in destroy
    if self._client_handle is None:
AttributeError: 'ActionClient' object has no attribute '_client_handle'
```

#### Additional information
1. The issue is not from Nav2 or geometry_msgs as they use the same releases that used to work in the previous builds (ubuntu builds of 20220211 do not manifest the issue)
2. At least 3 downstream projects are affected:
    - https://github.com/RobotWebTools/rosbridge_suite/issues/738
    - https://github.com/ros-planning/navigation2/issues/2942#issuecomment-1121234751
    - https://github.com/ros-planning/navigation2/issues/2942#issue-1228578420
3. Reverting the commit https://github.com/ros2/rosidl/commit/e9eea15c89b85094704a391ada5bb5c3b9688c0c and rebuilding rosidl and downstream packages fixes the issue. So I think that commit is to be reviewed.

hodgespodge, 2022-05-11:
I am also having this problem in a recent docker image build: `ImportError: /opt/ros/galactic/lib/libgazebo_msgs__rosidl_generator_c.so: undefined symbol: geometry_msgs__msg__Vector3__Sequence__are_equal`.

Running `sudo apt-get update ; sudo apt-get dist-upgrade` did fix the problem. 

Of course, I would prefer not to need to add `apt-get dist-upgrade` to my dockerfiles

clalancette, 2022-05-12:
> Of course, I would prefer not to need to add `apt-get dist-upgrade` to my dockerfiles

In general, we recommend that you do this upgrade in the dockerfiles, exactly to avoid this type of problem.

closed 2022-05-12 by m2-farzan