ros2/rosidl #686: C++ headers generated at build time with rosidl CLI

terrakuh, 2022-07-05:
## Question

Is it possible to compile at build time generated IDL-files using `rosidl_cli` to C++ headers and build a ROS service node with that?

### Setup

- OS: Ubuntu 22.04
- Compiler: GCC 12
- CMake: 3.22

#### CMakeLists.txt

```cmake
cmake_minimum_required(VERSION 3.20)

project(test)

find_package(rclcpp REQUIRED)
add_executable(my-test main.cpp my_service/srv/test__type_support.cpp)
target_link_libraries(my-test rclcpp::rclcpp)
target_include_directories(my-test PRIVATE "${CMAKE_CURRENT_SOURCE_DIR}")
```

#### main.cpp

```cpp
#include <rclcpp/rclcpp.hpp>

#include "my_service/srv/test.hpp"

inline void test(const std::shared_ptr<my_service::srv::Test::Request> request,
                              std::shared_ptr<my_service::srv::Test::Response> response)
{
  puts("Just a test.");
}

int main(int argc, char** argv)
{
  rclcpp::init(argc, argv);
  const auto node = rclcpp::Node::make_shared("test");

  const auto service_grip = node->create_service<my_service::srv::Test>("test", &test);

  std::cout << "ROS2 node up and running\n";
  rclcpp::spin(node);
  rclcpp::shutdown();
}
```

#### srv/Test.srv

```
string msg
---
string reply
```

My project structure is also attached here: [test.tar.gz](https://github.com/ros2/rosidl/files/9044487/test.tar.gz)

### Steps

Where the files reside I execute the following:

```sh
rosidl generate --type cpp --type-support cpp my_service srv/Test.srv
mv cpp my_service
cmake -S . -B build
cmake --build build
./build/my-test
```

Output:

```txt
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  could not create service: Type support not from this implementation. Got:
    Handle's typesupport identifier (rosidl_typesupport_cpp) is not supported by this library, at ./src/type_support_dispatch.hpp:111
    Could not load library libmy_service__rosidl_typesupport_fastrtps_cpp.so: dlopen error: libmy_service__rosidl_typesupport_fastrtps_cpp.so: cannot open shared object file: No such file or directory, at ./src/shared_library.c:99, at ./src/type_support_dispatch.hpp:76
while fetching it, at ./src/rmw_service.cpp:125, at ./src/rcl/service.c:124
Aborted (core dumped)
```

Rodario, 2022-07-19:
I also wondered about this. Would the cmake be the correct place for that tho? It would be pretty nice to just have a cmake function that compiles your srv / msg file and links the result to your project.

clalancette, 2022-08-04:
There is a design PR opened about this topic: https://github.com/ros2/design/pull/310 .  While we did expose some parts of the `rosidl` pipeline to the command-line, I don't believe we ever implemented all of the bits necessary to do what you'd like to do.

So this is future work.  That said, if you use the normal CMake mechanisms for packages and messages (like in https://docs.ros.org/en/rolling/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Cpp-Publisher-And-Subscriber.html), CMake should take care of all of this for you.