ros2/rosidl #699: LINK : fatal error LNK1104 when building on windows 10

GiladRobee, 2022-08-08:
## Bug report

LINK : fatal error LNK1104: cannot open file 'fastcdr-1.0.lib' [C:\ws_robee\build\robee_msg\robee_msg__rosidl_typesupport_fastrtps_c.vcxproj]
LINK : fatal error LNK1104: cannot open file 'fastcdr-1.0.lib' [C:\ws_robee\build\robee_msg\robee_msg__rosidl_typesupport_fastrtps_cpp.vcxproj]

**Required Info:**

- Operating System:
  - Windows 10 IOT
- Installation type:
  - binaries
- Version or commit hash:
  - ros2 humble
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue

```
colcon build --event-handlers console_cohesion+
```

#### Expected behavior
build complete successfully
#### Actual behavior

c:\ws_robee>colcon build --event-handlers console_cohesion+
Starting >>> robee_msg
--- output: robee_msg
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.17763.
-- Found ament_cmake: 1.3.2 (C:/dev/ros2_humble/share/ament_cmake/cmake)
-- Found builtin_interfaces: 1.2.0 (C:/dev/ros2_humble/share/builtin_interfaces/cmake)
-- Found rosidl_generator_c: 3.1.3 (C:/dev/ros2_humble/share/rosidl_generator_c/cmake)
-- Found rosidl_adapter: 3.1.3 (C:/dev/ros2_humble/share/rosidl_adapter/cmake)
-- Found rosidl_generator_cpp: 3.1.3 (C:/dev/ros2_humble/share/rosidl_generator_cpp/cmake)
-- Using all available rosidl_typesupport_c: rosidl_typesupport_fastrtps_c;rosidl_typesupport_introspection_c
-- Using all available rosidl_typesupport_cpp: rosidl_typesupport_fastrtps_cpp;rosidl_typesupport_introspection_cpp
-- Found std_msgs: 4.2.2 (C:/dev/ros2_humble/share/std_msgs/cmake)
-- Found geometry_msgs: 4.2.2 (C:/dev/ros2_humble/share/geometry_msgs/cmake)
-- Found rosidl_default_generators: 1.2.0 (C:/dev/ros2_humble/share/rosidl_default_generators/cmake)
-- Found ament_cmake_ros: 0.10.0 (C:/dev/ros2_humble/share/ament_cmake_ros/cmake)
-- Using all available rosidl_typesupport_c: rosidl_typesupport_fastrtps_c;rosidl_typesupport_introspection_c
-- Using all available rosidl_typesupport_cpp: rosidl_typesupport_fastrtps_cpp;rosidl_typesupport_introspection_cpp
-- Found python_cmake_module: 0.10.0 (C:/dev/ros2_humble/share/python_cmake_module/cmake)
-- Using PYTHON_EXECUTABLE: C:/Python38/python.exe
-- Using PYTHON_INCLUDE_DIRS: C:/Python38/include
-- Using PYTHON_LIBRARIES: optimized;C:/Python38/libs/python38.lib;debug;C:/Python38/libs/python38_d.lib
-- Using numpy include directory: C:\Python38\lib\site-packages\numpy\core\include
-- Configuring done
-- Generating done
-- Build files have been written to: C:/ws_robee/build/robee_msg
Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  running egg_info
  writing robee_msg.egg-info\PKG-INFO
  writing dependency_links to robee_msg.egg-info\dependency_links.txt
  writing top-level names to robee_msg.egg-info\top_level.txt
  reading manifest file 'robee_msg.egg-info\SOURCES.txt'
  writing manifest file 'robee_msg.egg-info\SOURCES.txt'
     Creating library C:/ws_robee/build/robee_msg/Release/robee_msg__rosidl_generator_c.lib and object C:/ws_robee/build/robee_msg/Release/robee_msg__rosidl_generator_c.exp
  robee_msg__rosidl_generator_c.vcxproj -> C:\ws_robee\build\robee_msg\Release\robee_msg__rosidl_generator_c.dll
     Creating library C:/ws_robee/build/robee_msg/Release/robee_msg__rosidl_typesupport_c.lib and object C:/ws_robee/build/robee_msg/Release/robee_msg__rosidl_typesupport_c.exp
  robee_msg__rosidl_typesupport_c.vcxproj -> C:\ws_robee\build\robee_msg\Release\robee_msg__rosidl_typesupport_c.dll
     Creating library C:/ws_robee/build/robee_msg/Release/robee_msg__rosidl_typesupport_cpp.lib and object C:/ws_robee/build/robee_msg/Release/robee_msg__rosidl_typesupport_cpp.exp
  robee_msg__rosidl_typesupport_cpp.vcxproj -> C:\ws_robee\build\robee_msg\Release\robee_msg__rosidl_typesupport_cpp.dll
LINK : fatal error LNK1104: cannot open file 'fastcdr-1.0.lib' [C:\ws_robee\build\robee_msg\robee_msg__rosidl_typesupport_fastrtps_c.vcxproj]
LINK : fatal error LNK1104: cannot open file 'fastcdr-1.0.lib' [C:\ws_robee\build\robee_msg\robee_msg__rosidl_typesupport_fastrtps_cpp.vcxproj]
     Creating library C:/ws_robee/build/robee_msg/Release/robee_msg__rosidl_typesupport_introspection_c.lib and object C:/ws_robee/build/robee_msg/Release/robee_msg__rosidl_typesupport_introspection_c.exp
  robee_msg__rosidl_typesupport_introspection_c.vcxproj -> C:\ws_robee\build\robee_msg\Release\robee_msg__rosidl_typesupport_introspection_c.dll
     Creating library C:/ws_robee/build/robee_msg/Release/robee_msg__rosidl_typesupport_introspection_cpp.lib and object C:/ws_robee/build/robee_msg/Release/robee_msg__rosidl_typesupport_introspection_cpp.exp
  robee_msg__rosidl_typesupport_introspection_cpp.vcxproj -> C:\ws_robee\build\robee_msg\Release\robee_msg__rosidl_typesupport_introspection_cpp.dll
---
Failed   <<< robee_msg [12.8s, exited with code 1]

Summary: 0 packages finished [13.5s]
  1 package failed: robee_msg
  4 packages not processed

#### Additional information
first build after the installation of ros2 humble. sourced the distro and ran all commands througth x86_84 terminal as an administor.

GiladRobee, 2022-08-10:
update:
there was an issue with the fact that i have downloaded the debugg version instead of the realse one.
switched and was partialy fixed.

closed 2022-08-10 by GiladRobee

InfoShellTex, 2023-06-26:
what did you do to switch to release? I believe I'm having the same issue trying to build ros2_dotnet. If possible, I would prefer an answer basically like your talking to a 5 year old lol. Though any extra input is better than nothing.

mjcarroll, 2023-06-27:
> what did you do to switch to release? I believe I'm having the same issue trying to build ros2_dotnet.

If you are using the fat binaries, just change which one you are downloading (eg from here https://github.com/ros2/ros2/releases/tag/release-humble-20230614)