ros2/rosidl #704: How to add __attribute__((packed)) to structs

mbUSC, 2022-09-20:
Is there an option to add `__attributes__((packed))` to C/C++ struct definitions generated from the .msg files?
I would like to avoid modifying them manually

clalancette, 2022-09-20:
No, there's currently not an option.

Could you describe your use case?

mbUSC, 2022-09-20:
Sure; my use case involves some simple communication with an embedded device over a serial connection.

The header mapping the message is shared between two different builds, which must use different compilers (this is because the device running ROS and the embedded device have different architectures).

The fact that padding is implementation-defined (i.e. depends on the compiler) can cause problems

clalancette, 2022-09-21:
OK, yeah, I figured it was something like that.

In general, we don't guarantee that the structures have any particular layout in memory.  They are really meant to be internal-to-ROS structures, and so their layout only needs to be consistent on the current machine (they'll get serialized before sending over the wire, and the receiving side will deserialize them).

Personally, I don't think we should expand their scope beyond that, so I would say we shouldn't add an option to pack them.  But maybe others think differently; does anyone from @ros2/team have an opinion here?

tfoote, 2022-10-11:
I don't think that we'd want this to be an option. We'd want it to be consistent for all users and expectations. I'm assuming that the use of this flag might have performance implications which is why the specification is left open to the compiler to take advantage of the architecture for optimizations. 

For communicating with an embedded device I would suggest that MicroROS might be a better approach to recommend rather than expecting to pass the in memory structure over the wire.

mbUSC, 2022-11-09:
Ah ok, that makes sense. Thank you for the explanation! :)