ros2/rosidl #705: deserialization fails when float is NaN

gitartpiano, 2022-09-21:
Example Python code generated by rosidl for float type field
(in /opt/ros/galactic/lib/python3.8/site-packages/sensor_msgs/msg/_battery_state.py):
```
    @capacity.setter
    def capacity(self, value):
        if __debug__:
            assert \
                isinstance(value, float), \
                "The 'capacity' field must be of type 'float'"
        self._capacity = value
```

The above assertion fails when the float value is ".nan".
```
File "/opt/ros/humble/local/lib/python3.10/dist-packages/sensor_msgs/msg/_battery_state.py", line 469, in capacity
    assert value >= -3.402823e+38 and value <= 3.402823e+38, \
AssertionError: The 'capacity' field must be a float in [-3.402823e+38, 3.402823e+38]
```

Here is an example message (which failed to de-serialize) captured by
"ros2 topic echo" command:
```
---
header:
  stamp:
    sec: 1663358019
    nanosec: 115042374
  frame_id: system
voltage: 161.0
temperature: 30.649999618530273
current: -4.269999980926514
charge: 304.20001220703125
capacity: .nan
design_capacity: 653.4000244140625
percentage: 0.7099999785423279
power_supply_status: 2
power_supply_health: 1
power_supply_technology: 2
present: true
cell_voltage:
- 3.5510001182556152
- 3.5930001735687256
cell_temperature:
- 29.700000762939453
- 31.600000381469727
location: ''
serial_number: '0'
```

Workaround was found to the problem; running Python with the -O flag, which sets __debug__ to false, disabling the assertion.
It would be better to fix the assertion, allowing the value to be NaN.

NaN is a valid value. According comment at https://github.com/ros2/common_interfaces/blob/rolling/sensor_msgs/msg/BatteryState.msg#L39
"(If unmeasured NaN)" .

Secondly: it would be nice to have support in the IDL for float type with NaN allowed and float type with Nan not allowed. Range checking would be nice to have too.

oysstu, 2024-04-26:
This was fixed in https://github.com/ros2/rosidl_python/pull/167 and backported to humble. I assume this issue can be closed as wontfix, unless a backport to galactic is on the table.

clalancette, 2024-04-26:
> I don't know what is on the table, but it would be correct to at least update the documentation of those releases which will not receive the backport to include notes about how to circumvent the issue.

Galactic is End-of-Life now, so we aren't going to update the code or the documentation for it.  We encourage you to move to a supported distribution, like Humble.

gitartpiano, 2024-04-26:
Yes, not touching End-of-Life-d releases is even more correct. Thanks for closing this.