ros2/rosidl #712: Build custom message failed."There are no rules to make targets,没有规则可制作目标"

noone3001, 2022-11-02:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - galactic
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
- Time.msg
```
uint32   sec
```
- CMakeLists.txt
```
cmake_minimum_required(VERSION 3.8)
project(ros2_interface)

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

# find dependencies
find_package(ament_cmake REQUIRED)
find_package(rosidl_default_generators REQUIRED)

rosidl_generate_interfaces(${PROJECT_NAME}
  "msg/Time.msg"
)
# uncomment the following section in order to fill in
# further dependencies manually.
# find_package(<dependency> REQUIRED)

if(BUILD_TESTING)
  find_package(ament_lint_auto REQUIRED)
  # the following line skips the linter which checks for copyrights
  # uncomment the line when a copyright and license is not present in all source files
  #set(ament_cmake_copyright_FOUND TRUE)
  # the following line skips cpplint (only works in a git repo)
  # uncomment the line when this package is not in a git repo
  #set(ament_cmake_cpplint_FOUND TRUE)
  ament_lint_auto_find_test_dependencies()
endif()

ament_package()
```
- package.xml
```
<?xml version="1.0"?>
<?xml-model href="http://download.ros.org/schema/package_format3.xsd" schematypens="http://www.w3.org/2001/XMLSchema"?>
<package format="3">
  <name>ros2_interface</name>
  <version>0.0.0</version>
  <description>TODO: Package description</description>
  <maintainer email="frank@todo.todo">frank</maintainer>
  <license>TODO: License declaration</license>

  <buildtool_depend>ament_cmake</buildtool_depend>
  <build_depend>rosidl_default_generators</build_depend>
  <exec_depend>rosidl_default_runtime</exec_depend>
  <member_of_group>rosidl_interface_packages</member_of_group>

  <test_depend>ament_lint_auto</test_depend>
  <test_depend>ament_lint_common</test_depend>

  <export>
    <build_type>ament_cmake</build_type>
  </export>
</package>
```

#### Expected behavior
```
Starting >>> ros2_interface
Finished <<< ros2_interface [0.65s]                  

Summary: 1 package finished [1.03s]
```
#### Actual behavior
```
Starting >>> ros2_interface
--- stderr: ros2_interface                             
make[1]: *** 没有规则可制作目标“ros2_interface__py/CMakeFiles/ros2_interface__py.dir/all”，由“CMakeFiles/ros2_interface__python.dir/all” 需求。 停止。
make[1]: *** 正在等待未完成的任务....
make: *** [Makefile:146：all] 错误 2
---
Failed   <<< ros2_interface [2.69s, exited with code 2]

Summary: 0 packages finished [3.09s]
  1 package failed: ros2_interface
  1 package had stderr output: ros2_interface
```
#### Additional information
- directory path
src/ros2_interface/msg/Time.msg

clalancette, 2022-11-17:
For what it is worth, this works just fine in my testing on Galactic here.  I guess we'll need more information about how your workspace is setup.

sloretz, 2023-10-13:
Thanks for the issue! Since Galactic is EOL I'll close this for now. If it continues to happen on a more recent version then we can reopen this.

closed 2023-10-13 by sloretz