ros2/rosidl #725: Coverity report CHECKED_RETURN issue

HKH347710, 2023-02-15:
## Bug report

**Required Info:**

- Operating System:
  - ubuntu20.04
- Installation type:
  - sudo apt install ros-foxy-desktop-full
- Version or commit hash:
  - Foxy
- DDS implementation:
  - default
- Client library (if applicable):
  - N/A
- Coverity
  - Coverity 2022.3.1 
#### Steps to reproduce issue

```
cov-configure --gcc
cov-build --dir analysis_dir colcon build
cov-analyze --dir analysis_dir --all --aggressiveness-level high
```

#### Actual behavior
Coverity report issues: unchecked return value (CHECKER_RETURN)
Issues file: <src workspace>/build/rslidar_msg/rosidl_typesupport_introspection_c/rslidar_msg/msg/detail/rslidar_packet__type_support.c
```
void RslidarScan__rosidl_typesupport_introspection_c__RslidarScan_init_function(
 void * message_memory, enum rosidl_runtime_c__message_initialization _init)
 {
   // TODO(karsten1987): initializers are not yet implemented for typesupport c
   // see https://github.com/ros2/ros2/issues/397
   (void) _init;
   	
CID xxx (#3 of 3): Unchecked return value (CHECKED_RETURN)
1. check_return: Calling rslidar_msg__msg__RslidarScan__init without checking return value (as is done elsewhere 3 out of 4 times).
   rslidar_msg__msg__RslidarScan__init(message_memory);
 }
```

#### Additional information
Source code analyzed:
  repo : https://github.com/RoboSense-LiDAR/rslidar_msg
  commit : 1a416d44c0132230adc781e8768954865f3b8c49

clalancette, 2023-02-23:
Thanks for reporting this bug.  Things may have changed in this area since Foxy; can you please try again with Humble or Rolling?

Further, we don't have coverity licenses, so if you could submit a PR to fix this, that would be helpful.

HKH347710, 2023-02-24:
Humble/Rolling has the same issue.

clalancette, 2023-02-24:
Thanks for the test, appreciated.