ros2/rosidl #749: Add non-default constructors to avoid initializing messages with dummy data

alsora, 2023-06-02:
## Feature request

ROS 2 C++ messages are implemented as data-structures with a default constructor.
Usually the workflow is
```
auto my_header = std_msgs::msg::Header();
my_header.frame_id = "hello";
```

This results in first zero-initializing the messages and then populating them again with the true data.
This has a non-negligible overhead in case of messages with large bounded members.

We propose to modify the IDL code to include a constructor that takes argument values, for example in the form of a brace-enclosed list
```
auto my_header = std_msgs::msg::Header({.frame_id = "hello"});
```

clalancette, 2023-06-15:
@alsora We think that the syntax there isn't supported until C++17 or C++20.  Can you check which version supports it?

clalancette, 2023-06-15:
Looking at it further, this looks to be a C++20 feature: https://en.cppreference.com/w/cpp/language/aggregate_initialization .

Also, it is possible that there don't need to be changes to the core to do this, and it would Just Work(tm) if we upgraded to C++20.

MiguelCompany, 2023-07-03:
I think that passing [this constant](https://github.com/ros2/rosidl/blob/187210e921c2a70390943ea87e8fcd90c27c14ba/rosidl_runtime_cpp/include/rosidl_runtime_cpp/message_initialization.hpp#L30) to the constructor and then assign all the fields in the message will serve the same purposes.

alsora, 2023-07-03:
@mauropasse did you test the `SKIP` mode?

mauropasse, 2023-07-04:
Yes, passing `rosidl_runtime_cpp::MessageInitialization::SKIP` as argument to the message constructor has the effect of not initialize anything. So for example, the cost of instantiating an `8MB` array message is the same as instantiating a message containing just a `bool`.

MiguelCompany, 2023-07-04:
It might be nice to have an overload of [the LoanedMessage constructor](https://github.com/ros2/rclcpp/blob/2e355e48498d7610534e590992c4dcff55a57f22/rclcpp/include/rclcpp/loaned_message.hpp#L59) that allows forwarding arguments (i.e. `rosidl_runtime_cpp::MessageInitialization`) to the message constructor when it is called [here](https://github.com/ros2/rclcpp/blob/2e355e48498d7610534e590992c4dcff55a57f22/rclcpp/include/rclcpp/loaned_message.hpp#L81), and perhaps also passing a similar argument to [this rmw function](https://github.com/ros2/rmw/blob/e523e4706ae6e20134fb835a91d18b74a134175e/rmw/include/rmw/rmw.h#L436) to indicate whether the borrowed memory shall be initialized or not.

jmachowinski, 2024-12-17:
> ## Feature request
> We propose to modify the IDL code to include a constructor that takes argument values, for example in the form of a brace-enclosed list
> 
> ```
> auto my_header = std_msgs::msg::Header({.frame_id = "hello"});
> ```

https://en.cppreference.com/w/cpp/language/aggregate_initialization

This is not possible, as this only works for aggregates.
The standard above states : 
<h4><span class="mw-headline" id="Aggregate">Aggregate</span></h4>
aggregate is one of the following types:

- ...
- class types that has
  - no user-declared constructors
  - ...
 
Therefore your proposal won't work, even if C++20 would be available.

@alsora @clalancette FYI.

Should we close this issue now ?

alsora, 2025-03-07:
Closing this.
Refer to https://github.com/ros2/rosidl/issues/842 for continuation

closed 2025-03-07 by alsora