ros2/rosidl #763: Unintended different in C/C++ APIs

allsey87, 2023-08-26:
I noticed that the return type for `resize_function` in [`rosidl_typesupport_introspection_c__MessageMember`](https://github.com/ros2/rosidl/blob/f47565cf5f0edfc535e5278503a6a0bce772f1e5/rosidl_typesupport_introspection_c/include/rosidl_typesupport_introspection_c/message_introspection.h#L78) is `bool` while in [`rosidl_typesupport_introspection_cpp::MessageMember`](https://github.com/ros2/rosidl/blob/f47565cf5f0edfc535e5278503a6a0bce772f1e5/rosidl_typesupport_introspection_cpp/include/rosidl_typesupport_introspection_cpp/message_introspection.hpp#L84C49-L84C49) it is `void`. Is this an oversight?

clalancette, 2023-08-28:
> I noticed that the return type for `resize_function` in [...]

I'm not 100% sure, but I think this is intended.  The C++ version, as implemented, can only fail by throwing an exception, so a `void` is appropriate.  The only way the C version can return an error is through a return code, so it returns a `bool`.

allsey87, 2023-08-29:
I see the logic there and will close this issue. Although this feels like an implementation detail leaking into the abstraction. My intuition would be that they should both return `bool` with the C++ variant catching `std::bad_alloc` and returning false...

closed 2023-08-29 by allsey87