ros2/rosidl #775: `char` type in Python messages is of type uint8_t not str.

arjo129, 2023-11-15:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04 and windows 10
- Installation type:
  - Binaries
- Version or commit hash:
  - Humble
- DDS implementation:
  - FastDDS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

The [documentation](https://docs.ros.org/en/humble/Concepts/Basic/About-Interfaces.html) page states that `char`s should corresponf to `builtins.str` but when I try to set a field of type `char` to a string in python like so: 
```
my_awesome_msg.char_field = "L"
```
I get the error
```
AssertionError: The 'my_awesome_msg.char_field' field must be of type 'int'
```

Is the documentation wrong or is this unexepected behaviour?

sloretz, 2023-12-15:
Thank you for the bug report! It seems like you're correct that the type should be a `str` field.

* ROS msg files to OMG IDL maps char to char: https://design.ros2.org/articles/mapping_dds_types.html
* OMG IDL char maps to python string with length 1: https://design.ros2.org/articles/idl_interface_definition.html

It seems like the bug is in `rosidl_adapter`. It maps ROS msg char to OMG IDL uint8: https://github.com/ros2/rosidl/blob/5d8191e07627a8c394e42690d6a090d2f63c016a/rosidl_adapter/rosidl_adapter/msg/__init__.py#L46 . 

Would you be willing to open a PR with a fix on the rolling branch? I don't think we could backport it, but we could fix it going forward.

arjo129, 2024-06-07:
I think updating the documentation is sufficient. Changing this is not worth the effort as users who use this bug will likely have to go through  a very painful process of updating their code base for very little value-add.

closed 2024-06-07 by arjo129