ros2/rosidl #784: rosidl_adapter

Darshi18, 2024-02-01:
I am facing an issue while building MoveIt on ROS [HUMBLE] with Python 3.10.12. The error is related to the rosidl_adapter module and seems to involve an attribute error with the em module. Here’s the relevant error message:

CMake Error at /opt/ros/humble/share/rosidl_adapter/cmake/rosidl_adapt_interfaces.cmake:59 (message):
execute_process(/usr/local/bin/python3.7 -m rosidl_adapter --package-name
controller_manager_msgs --arguments-file
/home/ubuntu/build/controller_manager_msgs/rosidl_adapter__arguments__controller_manager_msgs.json
–output-dir
/home/ubuntu/build/controller_manager_msgs/rosidl_adapter/controller_manager_msgs
–output-file
/home/ubuntu/build/controller_manager_msgs/rosidl_adapter/controller_manager_msgs.idls)
returned error code 1:

AttributeError processing template ‘msg.idl.em’

Traceback (most recent call last):

File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/resource/__init__.py", line 48, in evaluate_template
  _interpreter = em.Interpreter(

AttributeError: module ‘em’ has no attribute ‘Interpreter’

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

File "/usr/local/lib/python3.7/runpy.py", line 193, in _run_module_as_main
  "__main__", mod_spec)
File "/usr/local/lib/python3.7/runpy.py", line 85, in _run_code
  exec(code, run_globals)
File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/__main__.py", line 19, in <module>
  sys.exit(main())
File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/main.py", line 55, in main
  pathlib.Path(relative_path), output_dir)
File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/__init__.py", line 20, in convert_to_idl
  package_dir, package_name, interface_file, output_dir / 'msg')
File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/msg/__init__.py", line 39, in convert_msg_to_idl
  expand_template('msg.idl.em', data, output_file, encoding='iso-8859-1')
File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/resource/__init__.py", line 23, in expand_template
  content = evaluate_template(template_name, data)
File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/resource/__init__.py", line 69, in evaluate_template
  _interpreter.shutdown()

AttributeError: ‘NoneType’ object has no attribute ‘shutdown’

Call Stack (most recent call first):
/opt/ros/humble/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:130 (rosidl_adapt_interfaces)
CMakeLists.txt:28 (rosidl_generate_interfaces)

**Details:**

    ROS Distribution: [HUMBLE]
    MoveIt Version: [MOVEIT2]
    Python Version: 3.10.12
    rosidl_adapter Version: 3.1.5
    em Version :  0.4.0

clalancette, 2024-02-01:
> em Version :  0.4.0

This is the problem.  There are two confusingly named Python projects, one called `em` and one called `empy`.  Even more confusing, they both use the `em` namespace.  ROS 2 depends on `empy` (version 3.3.4), but you have `em` installed.

In the future, please carefully follow the [installation instructions](http://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Development-Setup.html), which should have installed the correct thing for you.

Darshi18, 2024-02-02:
okay but i also have empy(version 3.3.4) so in order to resolve the issue i need to uninstall em module ?

clalancette, 2024-02-02:
> okay but i also have empy(version 3.3.4) so in order to resolve the issue i need to uninstall em module ?

Yes, that should do it.