ros2/rosidl #786: Add [[nodiscard]] attribute to C++ headers that return values, such as operators [Jazzy]

Ryanf55, 2024-02-26:
## Feature request

#### Feature description

Getters and operators could have the [[nodiscard]] attribute to prevent bugs.

#### Implementation considerations

See `/opt/ros/humble/include/geometry_msgs/geometry_msgs/msg/detail/point__struct.hpp`. This is ABI breaking because it increases the strictness of the API.

Before:
```c++
  bool operator!=(const Point_ & other) const
  {
    return !this->operator==(other);
  }
  

```

Proposed:
```
  [[nodiscard]] bool operator!=(const Point_ & other) const
  {
    return !this->operator==(other);
  }
```

Effect:

Using the `!=` operator without assigning the value would then issue a compile warning.

#### Related

Same as https://github.com/gazebosim/sdformat/issues/1358

wjwwood, 2024-03-07:
Reviewing from our issue triaging meeting, sounds like a nice improvement, help is appreciated.

jmachowinski, 2024-11-26:
Is this issue fixed and can be closed ?

closed 2024-11-26 by clalancette