ros2/rosidl #787: Support min/max attribute on primitive types

Ryanf55, 2024-02-28:
## Feature request

Can ROSIDL have a way to specify limits for primates in messages?

#### Feature description

Currently, in geographic_msgs/msg/GeoPoint, it has this (quite long) and confusing comment on the values.
```
# Geographic point, using the WGS 84 reference ellipsoid.

# Latitude [degrees]. Positive is north of equator; negative is south
# (-90 <= latitude <= +90).
float64 latitude

# Longitude [degrees]. Positive is east of prime meridian; negative is
# west (-180 <= longitude <= +180). At the poles, latitude is -90 or
# +90, and longitude is irrelevant, but must be in range.
float64 longitude

# Altitude [m]. Positive is above the WGS 84 ellipsoid (NaN if unspecified).
float64 altitude
```

As a user of this message in production code, I must perform runtime validation on any incoming GeoPoint messages to verify the latitude and longitude fields are valid. This puts the load on users of ROS 2 messages, and I have to make a new package to validate open source messages.

In an ideal world, we could leverage runtime bounds checking, and codify it in the DDS IDL with `@min` and `@max` attributes.

Instead, I've submitted a PR to add runtime validation to the specific repo because ROSIDL doesn't support primitive bounds limits. A python user of geographic_msgs thus does not benefit from my contribution. 

#### Proposal

Add support for `@` attributes that are directly above the primitive.

```
# Geographic point, using the WGS 84 reference ellipsoid.

# Latitude [degrees]. Positive is north of equator; negative is south
# (-90 <= latitude <= +90).
# @min -90.0
# @max 90.0
float64 latitude

# Longitude [degrees]. Positive is east of prime meridian; negative is
# west (-180 <= longitude <= +180). At the poles, latitude is -90 or
# +90, and longitude is irrelevant, but must be in range.
# @min -180.0
# @max 180.0
float64 longitude

# Altitude [m]. Positive is above the WGS 84 ellipsoid (NaN if unspecified).
float64 altitude
```

#### Implementation considerations

The alternative is a very large PR like this to message repos to perform bounds checking:
https://github.com/ros-geographic-info/geographic_info/pull/56

eProsima needs to add support:
https://fast-dds.docs.eprosima.com/en/latest/fastddsgen/dataTypes/dataTypes.html#primitive-types

cycloneDDS needs to add support:
https://cyclonedds.io/content/guides/supported-idl.html

If I were a DDS vendor, I would add this as part of the `setter` to the value. Because ROSIDL exposes the data members directly as public attributes (at least in C++), I have no idea how ROSIDL can validate them. Open to ideas.

christophebedard, 2024-02-29:
As you mentioned, OMG IDL has the `@range`/@min`/`@max` attributes (section 8.3.3: _Group of Annotations: Units and Ranges_ of version 4.2).

> I have no idea how ROSIDL can validate them.

We could just add support for the `@range`/@min`/`@max` annotations to `rosidl` and then let the middleware implementations do the rest. `rosidl` doesn't really need to deal with it.

wjwwood, 2024-03-07:
From our issue triaging meeting, the consensus was that adding support for these keywords would just be part of properly supporting this change, and the scope of it is large enough that if we were to go down this path, a REP should be made. To that end, I'll include more notes below and then close this as answered because to move forward someone needs to push for this by making a REP about it.

---

A few things to consider:

- adding support in ROSIDL would just mean it's translated into the correct annotations in the OMG IDL (all ROSIDL files get translated into OMG IDL before being processed by our type support pipeline)
- we only support a subset of OMG IDL in our type support pipeline, so we'd have to add these annotations to that subset (see: https://design.ros2.org/articles/idl_interface_definition.html)
- if we added this support most of the DDS implementations would need to be updated to support it (as you pointed out the two main open source ones currently do not) and other rmw implementations would also need to support it (e.g. the up coming rmw_zenoh project)
- once the middleware supported it, we have to figure out how to handle when these constraints are violated, consider what happens when a remote node (maybe an old bag file) sends a value that's out of range, how does that present itself on the receiving side? is the message dropped? is there a warning? does the node throw and crash? is there a new callback or something for when a subscription gets an invalid message? etc...
- on the publishing side, is this enforced only at the middleware layer? does it throw some how on publish? do we add the checking only on assignment or on demand with the ROS message struct? if so how since we allow member access to the data fields?

I believe that captured most of the concerns (that would need to be addressed in a REP about this), but there may be others.

Also mentioned in the meeting:

- The type adaption feature in C++ would allow us to have a new open source package that just provides a bounds checking type that you could use for publishing and subscribing, using the `geographic_msgs/msg/GeoPoint` on the wire only or in languages that do not support that (e.g. Python), see: https://github.com/ros2/examples/blob/c370d7e7eb6570705794f4f166cec626ba4870b2/rclcpp/topics/minimal_publisher/member_function_with_type_adapter.cpp#L63-L94
- You could just have validating functions in another open source package, like `geographic_msgs_cpp` or `geographic_msgs_utils` or something like that, which we have in other places, e.g. there's a point cloud iterator in the `sensor_msgs` package: https://github.com/ros2/common_interfaces/blob/rolling/sensor_msgs/include/sensor_msgs/point_cloud2_iterator.hpp (though you should really not put them in the same package as the messages, these are here from a long time ago before we recommended splitting messages and code, which lead to us having a separate `sensor_msgs_py` which is kind of unfortunate)

---

As I mentioned above, I'll close this for now, but feel free to continue the discussion and/or ask for it to be reopened if you think that's necessary.

closed 2024-03-07 by wjwwood