ros2/rosidl #807: Error while building for Windows - error count exceeds 100

RParkerE, 2024-05-02:
## Bug report

**Required Info:**

- Operating System:
Windows 10 (10.0.19045 Build 19045)
- Installation type:
Source
- Version or commit hash:
[`995917e`](https://github.com/ros2/rosidl/commit/995917e9ce14d17821c04bf28d5a092111537842) (iron branch)
- DDS implementation:
Eclipse Cyclone DDS
- Client library (if applicable):
rclpy

#### Steps to reproduce issue
Follow ROS2 Windows build instructions. (with fix [`#96`](https://github.com/ros/resource_retriever/pull/96) for resource_retriever)

#### Expected behavior
Build succeeded

#### Actual behavior
Error when building `rosidl_typesupport_introspection_tests`
Failing with `fatal error C1003: error count exceeds 100`

#### Additional information
I can provide the full output of the log file, but since it is over 2k lines long, for brevity, here is just the first 50 lines:
```
Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  rosidl_typesupport_introspection_tests__rosidl_generator_c.vcxproj -> C:\iron\build\rosidl_typesupport_introspection_tests\Release\rosidl_typesupport_introspection_tests__rosidl_generator_c.dll
  rosidl_typesupport_introspection_tests__rosidl_typesupport_c.vcxproj -> C:\iron\build\rosidl_typesupport_introspection_tests\Release\rosidl_typesupport_introspection_tests__rosidl_typesupport_c.dll
  rosidl_typesupport_introspection_tests__rosidl_typesupport_cpp.vcxproj -> C:\iron\build\rosidl_typesupport_introspection_tests\Release\rosidl_typesupport_introspection_tests__rosidl_typesupport_cpp.dll
  rosidl_typesupport_introspection_tests__rosidl_typesupport_fastrtps_c.vcxproj -> C:\iron\build\rosidl_typesupport_introspection_tests\Release\rosidl_typesupport_introspection_tests__rosidl_typesupport_fastrtps_c.dll
  rosidl_typesupport_introspection_tests__rosidl_typesupport_fastrtps_cpp.vcxproj -> C:\iron\build\rosidl_typesupport_introspection_tests\Release\rosidl_typesupport_introspection_tests__rosidl_typesupport_fastrtps_cpp.dll
  rosidl_typesupport_introspection_tests__rosidl_typesupport_introspection_c.vcxproj -> C:\iron\build\rosidl_typesupport_introspection_tests\Release\rosidl_typesupport_introspection_tests__rosidl_typesupport_introspection_c.dll
  rosidl_typesupport_introspection_tests__rosidl_typesupport_introspection_cpp.vcxproj -> C:\iron\build\rosidl_typesupport_introspection_tests\Release\rosidl_typesupport_introspection_tests__rosidl_typesupport_introspection_cpp.dll
  gtest.vcxproj -> C:\iron\build\rosidl_typesupport_introspection_tests\gtest\Release\gtest.lib
  gtest_main.vcxproj -> C:\iron\build\rosidl_typesupport_introspection_tests\gtest\Release\gtest_main.lib
  test_arrays_message_introspection.cpp
C:\iron\build\rosidl_typesupport_introspection_tests\rosidl_generator_c\rosidl_typesupport_introspection_tests/msg/detail/basic_types__struct.h(1,1): warning C4821: Unable to determine Unicode encoding type, please save the file with signature (BOM) [C:\iron\build\rosidl_typesupport_introspection_tests\test_arrays_message_introspection.vcxproj]
C:\iron\build\rosidl_typesupport_introspection_tests\rosidl_generator_c\rosidl_typesupport_introspection_tests/msg/detail/constants__struct.h(1,1): warning C4821: Unable to determine Unicode encoding type, please save the file with signature (BOM) [C:\iron\build\rosidl_typesupport_introspection_tests\test_arrays_message_introspection.vcxproj]
C:\iron\build\rosidl_typesupport_introspection_tests\rosidl_generator_c\rosidl_typesupport_introspection_tests/msg/detail/defaults__struct.h(1,1): warning C4821: Unable to determine Unicode encoding type, please save the file with signature (BOM) [C:\iron\build\rosidl_typesupport_introspection_tests\test_arrays_message_introspection.vcxproj]
C:\iron\build\rosidl_typesupport_introspection_tests\rosidl_generator_c\rosidl_typesupport_introspection_tests/msg/detail/arrays__struct.h(50,59): error C3646: 'basic_types_values': unknown override specifier [C:\iron\build\rosidl_typesupport_introspection_tests\test_arrays_message_introspection.vcxproj]
C:\iron\build\rosidl_typesupport_introspection_tests\rosidl_generator_c\rosidl_typesupport_introspection_tests/msg/detail/arrays__struct.h(50,77): error C2143: syntax error: missing ',' before '[' [C:\iron\build\rosidl_typesupport_introspection_tests\test_arrays_message_introspection.vcxproj]
C:\iron\build\rosidl_typesupport_introspection_tests\rosidl_generator_c\rosidl_typesupport_introspection_tests/msg/detail/arrays__struct.h(50,80): error C2238: unexpected token(s) preceding ';' [C:\iron\build\rosidl_typesupport_introspection_tests\test_arrays_message_introspection.vcxproj]
C:\iron\build\rosidl_typesupport_introspection_tests\rosidl_generator_c\rosidl_typesupport_introspection_tests/msg/detail/arrays__struct.h(51,58): error C3646: 'constants_values': unknown override specifier [C:\iron\build\rosidl_typesupport_introspection_tests\test_arrays_message_introspection.vcxproj]
C:\iron\build\rosidl_typesupport_introspection_tests\rosidl_generator_c\rosidl_typesupport_introspection_tests/msg/detail/arrays__struct.h(51,74): error C2143: syntax error: missing ',' before '[' [C:\iron\build\rosidl_typesupport_introspection_tests\test_arrays_message_introspection.vcxproj]
C:\iron\build\rosidl_typesupport_introspection_tests\rosidl_generator_c\rosidl_typesupport_introspection_tests/msg/detail/arrays__struct.h(51,77): error C2238: unexpected token(s) preceding ';' [C:\iron\build\rosidl_typesupport_introspection_tests\test_arrays_message_introspection.vcxproj]
C:\iron\build\rosidl_typesupport_introspection_tests\rosidl_generator_c\rosidl_typesupport_introspection_tests/msg/detail/arrays__struct.h(52,57): error C3646: 'defaults_values': unknown override specifier [C:\iron\build\rosidl_typesupport_introspection_tests\test_arrays_message_introspection.vcxproj]
C:\iron\build\rosidl_typesupport_introspection_tests\rosidl_generator_c\rosidl_typesupport_introspection_tests/msg/detail/arrays__struct.h(52,72): error C2143: syntax error: missing ',' before '[' [C:\iron\build\rosidl_typesupport_introspection_tests\test_arrays_message_introspection.vcxproj]
C:\iron\build\rosidl_typesupport_introspection_tests\rosidl_generator_c\rosidl_typesupport_introspection_tests/msg/detail/arrays__struct.h(52,75): error C2238: unexpected token(s) preceding ';' [C:\iron\build\rosidl_typesupport_introspection_tests\test_arrays_message_introspection.vcxproj]
C:\iron\build\rosidl_typesupport_introspection_tests\rosidl_generator_c\rosidl_typesupport_introspection_tests/msg/detail/basic_types__struct.h(1,1): warning C4821: Unable to determine Unicode encoding type, please save the file with signature (BOM) [C:\iron\build\rosidl_typesupport_introspection_tests\test_arrays_message_introspection.vcxproj]
C:\iron\build\rosidl_typesupport_introspection_tests\rosidl_generator_c\rosidl_typesupport_introspection_tests/msg/detail/basic_types__struct.h(1,1): warning C4821: Unable to determine Unicode encoding type, please save the file with signature (BOM) [C:\iron\build\rosidl_typesupport_introspection_tests\test_arrays_message_introspection.vcxproj]
C:\iron\build\rosidl_typesupport_introspection_tests\rosidl_generator_c\rosidl_typesupport_introspection_tests/msg/detail/basic_types__functions.h(42,63): error C2065: 'rosidl_typesupport_introspection_tests__msg__BasicTypes': undeclared identifier [C:\iron\build\rosidl_typesupport_introspection_tests\test_arrays_message_introspection.vcxproj]
C:\iron\build\rosidl_typesupport_introspection_tests\rosidl_generator_c\rosidl_typesupport_introspection_tests/msg/detail/basic_types__functions.h(42,121): error C2065: 'msg': undeclared identifier [C:\iron\build\rosidl_typesupport_introspection_tests\test_arrays_message_introspection.vcxproj]
C:\iron\build\rosidl_typesupport_introspection_tests\rosidl_generator_c\rosidl_typesupport_introspection_tests/msg/detail/basic_types__functions.h(42,124): error C2491: 'rosidl_typesupport_introspection_tests__msg__BasicTypes__init': definition of dllimport data not allowed [C:\iron\build\rosidl_typesupport_introspection_tests\test_arrays_message_introspection.vcxproj]
C:\iron\build\rosidl_typesupport_introspection_tests\rosidl_generator_c\rosidl_typesupport_introspection_tests/msg/detail/basic_types__functions.h(50,63): error C2065: 'rosidl_typesupport_introspection_tests__msg__BasicTypes': undeclared identifier [C:\iron\build\rosidl_typesupport_introspection_tests\test_arrays_message_introspection.vcxproj]
[... cut ...]
C:\iron\build\rosidl_typesupport_introspection_tests\rosidl_generator_c\rosidl_typesupport_introspection_tests/msg/detail/basic_types__functions.h(139,155): error C2146: syntax error: missing ')' before identifier 'size' [C:\iron\build\rosidl_typesupport_introspection_tests\test_arrays_message_introspection.vcxproj]
C:\iron\build\rosidl_typesupport_introspection_tests\rosidl_generator_c\rosidl_typesupport_introspection_tests/msg/detail/basic_types__functions.h(151,73): error C2065: 'rosidl_typesupport_introspection_tests__msg__BasicTypes__Sequence': undeclared identifier [C:\iron\build\rosidl_typesupport_introspection_tests\test_arrays_message_introspection.vcxproj]
```

RParkerE, 2024-05-03:
Some more information: I decided to run `colcon build --merge-install --continue-on-error`, to see if there were any other issues preventing ros from building, and it looks like only `rosidl_typesupport_introspection_tests` is failing.

```
Summary: 358 packages finished [55min 19s]
  1 package failed: rosidl_typesupport_introspection_tests
  48 packages had stderr output: action_tutorials_interfaces actionlib_msgs composition composition_interfaces demo_nodes_cpp demo_nodes_cpp_native diagnostic_msgs example_interfaces geometry_msgs image_tools intra_process_demo laser_geometry libstatistics_collector lifecycle lifecycle_msgs logging_demo map_msgs message_filters nav_msgs pendulum_msgs rcl rcl_interfaces rclcpp rmw_connextdds_common rmw_dds_common robot_state_publisher ros2cli_test_interfaces rosbag2_interfaces rosbag2_test_msgdefs rosgraph_msgs rqt_py_common sensor_msgs shape_msgs statistics_msgs std_msgs std_srvs stereo_msgs test_cli test_cli_remapping test_communication test_msgs test_rclcpp test_security test_tf2 tf2_msgs trajectory_msgs turtlesim visualization_msgs
WNDPROC return value cannot be converted to LRESULT
TypeError: WPARAM is simple, so must be an int object (got NoneType)
```

sloretz, 2024-05-16:
It's strange that VS Code couldn't detect the file encoding.

* What version of VS Code do you have installed?
* Would you be willing to upload an archive with the folder `C:\iron\build\rosidl_typesupport_introspection_tests\rosidl_generator_c\rosidl_typesupport_introspection_tests/msg/detail/`?
* Would you be willing to see if the same issue happens with ROS Rolling?

knmcguire, 2025-02-24:
I've tried it out on Rolling (with the [PR of new pixi instructions](https://github.com/ros2/ros2_documentation/pull/4989) for windows) and confirm that this issue still exist.

It seems this issue of colcon build is related but currently digging into it if it actually works: https://github.com/colcon/colcon-notification/issues/64

knmcguire, 2025-02-24:
Here is the std_out of that error, but I think I found the issue... The name of the path is too long. Maximum is 260 characters and the path it's trying to write to is 269 characters. 

This max length path limitation that windows has it just really annoying. Currently trying out if there is any better way out of this now.

```
C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\MSBuild\Microsoft\VC\v160\Microsoft.CppBuild.targets(382,5): error MSB3491: Could not write lines to file "rosidl_typesupport_introspection_tests__rosidl_typesupport_introspection_cpp.dir\Release\rosidl_t.E2569E3A.tlog\rosidl_typesupport_introspection_tests__rosidl_typesupport_introspection_cpp.lastbuildstate". Path: rosidl_typesupport_introspection_tests__rosidl_typesupport_introspection_cpp.dir\Release\rosidl_t.E2569E3A.tlog\rosidl_typesupport_introspection_tests__rosidl_typesupport_introspection_cpp.lastbuildstate exceeds the OS max path limit. The fully qualified file name must be less than 260 characters. [C:\dev\rolling\build\rosidl_typesupport_introspection_tests\rosidl_typesupport_introspection_tests__rosidl_typesupport_introspection_cpp.vcxproj]
```
[stdout_stderr.log](https://github.com/user-attachments/files/18938624/stdout_stderr.log)

knmcguire, 2025-02-24:
Hmm... I see now that the original issue had probably a different issue and it's maybe not due to the path length 

Sorry for for that. I can delete my previous comments and add those to another issue, perhaps the  https://github.com/colcon/colcon-notification/issues/64 ? It will keep the issue clean but I know that some projects don't like deleting history.