ros2/rosidl #819: `rosidl` type support packages not found

alberthli, 2024-07-18:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04 (Dockerized)
- Installation type:
  - humble via pixi (robostack)
- DDS implementation:
  - both Fast-RTPS and Cyclone DDS
- Client library (if applicable):
  - both rclpy and rclcpp

This issue is cross-posted from here: https://github.com/prefix-dev/pixi/issues/1635. However, since `pixi` is such a specific tool, I was hoping to get any additional possible insight here as well. If it's better for organization, I'm happy to close this issue and consolidate the discussion to the other thread.

The full text of the original issue is pasted here for reference.

# Original Issue
### Reproducible example

Tl;dr: **Our issue arises when trying to boot up a ROS stack in a pixi shell. When using pixi, rosidl type support packages are not found. However, when we manually install ROS pre-compiled binaries in a Docker container without using pixi, the packages are found and our stack boots up correctly.**

We have not found an easily-reproducible minimal example. The issue can be reproduced in the following debugging branch of our project repo with the commit hash `46f503cf2ba7068b2b576e5540eb3ab17cea7f10`: [https://github.com/Caltech-AMBER/obelisk/tree/debug-docker-rosidl](https://github.com/Caltech-AMBER/obelisk/tree/debug-docker-rosidl).

Assuming you already have required system deps and Docker, run our setup script with the following flags in the root of the cloned repo on the `debug-docker-rosidl` branch:
```
source dev_setup.sh --skip-docker --no-cyclone-perf
```
The only thing this does is set an environment variable called `OBELISK_ROOT` to the directory where this repository has been cloned and creates a `.env` file containing other environment variables useful for dockerizing our dev setup. We run Docker for isolation and run pixi inside of it. To build and enter the container, run
```
cd docker
docker compose -f docker-compose-no-gpu.yml run --build obelisk
```
Then, to enter the `dev` environment, run
```
pixi shell -e dev
```
We build some custom messages through some pixi task, then the rest of the ROS2 packages.
```
pixi run messages-build
pixi run source-obelisk
```
Then, we boot up our stack using an alias installed in the Docker container:
```
obk-launch config_file_path=dummy_cpp.yaml device_name=onboard
```
This will spawn a simulation, but we see a runtime error in the terminal:
```
[obelisk_mujoco_robot-5] [rcutils|error_handling.c:65] an error string (message, file name, or formatted message) will be truncated
[obelisk_mujoco_robot-5] 
[obelisk_mujoco_robot-5] >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
[obelisk_mujoco_robot-5] This error state is being overwritten:
[obelisk_mujoco_robot-5] 
[obelisk_mujoco_robot-5]   'Type support not from this implementation. Got:
[obelisk_mujoco_robot-5]     Handle's typesupport identifier (rosidl_typesupport_cpp) is not supported by this library, at /opt/conda/build_artifacts/ros-humble-rosidl-typesupport-cpp-0_1675687196536/work/ros-humble-rosidl-typesupport-cpp/src/work/src/type_support_dispatch.hpp:111
[obelisk_mujoco_robot-5]     Could not load library libobelisk_sensor_msgs__rosidl_typesupport_introspection_cpp.so: dlopen error: libstd_msgs__rosidl_typesupport_introspection_cpp.so: cannot open shared object file: No such file or directory, at /opt/conda/build_artifacts/ros-humble-rcutils-0_1675685175987/work/ros-humble-rcutils/src/work/src/shared_library.c:99, at /opt/conda/build_artifacts/ros-humble-rosidl-typesupport-cpp-0_1675687196536/work/ros-humble-rosidl-typesupport-cpp/s, at /opt/conda/build_artifacts/ros-humble-rmw-cyclonedds-cpp-0_1675688132292/work/ros-humble-rmw-cyclonedds-cpp/src/work/src/rmw_node.cpp:1958'
[obelisk_mujoco_robot-5] 
[obelisk_mujoco_robot-5] with this new error message:
[obelisk_mujoco_robot-5] 
[obelisk_mujoco_robot-5]   'type_support is null, at /opt/conda/build_artifacts/ros-humble-rmw-cyclonedds-cpp-0_1675688132292/work/ros-humble-rmw-cyclonedds-cpp/src/work/src/rmw_node.cpp:2277'
[obelisk_mujoco_robot-5] 
[obelisk_mujoco_robot-5] rcutils_reset_error() should be called after error handling to avoid this.
[obelisk_mujoco_robot-5] <<<
```

### Issue description

Here, we summarize a few things we tested. Each test is independent from each other.

### Test 1: Using only Docker without pixi
[... 164 lines cut ...]

### Other Observations
We noticed that it's looking for `rosidl_typesupport` in `/opt/conda/...`, which we don't have in the container:
```
Handle's typesupport identifier (rosidl_typesupport_cpp) is not supported by this library, at /opt/conda/build_artifacts/ros-humble-rosidl-typesupport-cpp-0_1675687196536/work/ros-humble-rosidl-typesupport-cpp/src/work/src/type_support_dispatch.hpp:111
```
It also tries to look there for a `.so` file:
```
Could not load library libobelisk_sensor_msgs__rosidl_typesupport_fastrtps_cpp.so: dlopen error: libstd_msgs__rosidl_typesupport_fastrtps_cpp.so: cannot open shared object file: No such file or directory, at /opt/conda/build_artifacts/ros-humble-rcutils-0_1675685175987/work/ros-humble-rcutils/src/work/src/shared_library.c:99, at /opt/conda/build_artifacts/ros-humble-rosidl-typesupport-cpp-0_1675687196536/work/ros-humble-rosidl-typesupport-cpp/src/work/sr, at /opt/conda/build_artifacts/ros-humble-rcl-0_1675689729790/work/ros-humble-rcl/src/work/src/rcl/subscription.c:108'
```
However, we can find this file successfully in the directory `$OBELISK_ROOT/obelisk_ws/install/obelisk_sensor_msgs/lib`
```
# this shows a bunch of .so files
ls $OBELISK_ROOT/obelisk_ws/install/obelisk_sensor_msgs/lib

# outputs the below
libobelisk_sensor_msgs__rosidl_generator_c.so    libobelisk_sensor_msgs__rosidl_typesupport_cpp.so           libobelisk_sensor_msgs__rosidl_typesupport_introspection_c.so
libobelisk_sensor_msgs__rosidl_generator_py.so   libobelisk_sensor_msgs__rosidl_typesupport_fastrtps_c.so    libobelisk_sensor_msgs__rosidl_typesupport_introspection_cpp.so
libobelisk_sensor_msgs__rosidl_typesupport_c.so  libobelisk_sensor_msgs__rosidl_typesupport_fastrtps_cpp.so  python3.11
```
This suggests that we may be able to configure something slightly different such that these files are found.

### Expected behavior

Our ROS stack should boot up without issue.

traversaro, 2024-07-19:
To clarify, I also tried to use the instructions you provided for run the installation with docker, but executing:

~~~
source dev_setup.sh --skip-docker --no-cyclone-perf
~~~

tries to install some tools (`uv`) globally, so I just halted it to avoid the script messing with my environment.

alberthli, 2024-07-19:
Closing this issue in favor of activity on the other thread.

closed 2024-07-19 by alberthli

traversaro, 2024-07-20:
> To clarify, I also tried to use the instructions you provided for run the installation with docker, but executing:
> 
> ```
> source dev_setup.sh --skip-docker --no-cyclone-perf
> ```
> 
> tries to install some tools (`uv`) globally, so I just halted it to avoid the script messing with my environment.
> 
> It also tried to clutter my `.bashrc`, it would be ideal if you avoid doing that while presenting a way to reproduce your problem.

My bad, that was an answer for the other thread, not sure how it ended up here.