ros2/rosidl #822: rosidl_generator_cpp build fails on humble

christophfroehlich, 2024-08-21:
## Bug report
(Since yesterday?) the source build of the humble branch fails.
```
Starting >>> rosidl_generator_cpp
--- stderr: rosidl_generator_cpp                               
In file included from /workspaces/ros2_humble_ws/src/rosidl/rosidl_generator_cpp/test/test_msg_initialization.cpp:21:
/workspaces/ros2_humble_ws/build/rosidl_generator_cpp/rosidl_generator_cpp/rosidl_generator_cpp/msg/defaults.hpp:10:10: fatal error: rosidl_generator_cpp/msg/detail/defaults__type_support.hpp: No such file or directory
   10 | #include "rosidl_generator_cpp/msg/detail/defaults__type_support.hpp"
```

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - src
- Version or commit hash:
```
$ git rev-parse HEAD
6a18bbfa1fe80bc5474b19bbda50991f1ac9546b
```

#### Steps to reproduce issue
Checkout humble branch, rosdep, colcon build

clalancette, 2024-08-21:
Yeah, we merged in #778 , which also requires https://github.com/ros2/rosidl_typesupport/pull/149 .

christophfroehlich, 2024-08-23:
Hm, it still fails with the same error today? (everything built from source: 
https://github.com/ros-controls/ros2_controllers/actions/runs/10519339357/job/29146541748#step:9:9906)
Maybe a wrong branch in a repos file? Was there any involved repo branched for humble recently?

clalancette, 2024-08-24:
All of the CI jobs are happy: https://build.ros2.org/view/Hci/

And it built locally just fine for me.  Without looking at it closer, my best guess is that not everything is built from source in those GitHub actions.

christophfroehlich, 2024-08-24:
Thanks for having a look, maybe this is just an issue with caching of build artifacts in the workflow.

christophfroehlich, 2024-08-24:
I crawled through the logs, the correct branches are checked out:
```
2024-08-24T03:16:23.9787714Z === src/ros2/rosidl (git) ===
2024-08-24T03:16:23.9788657Z commit 6a18bbfa1fe80bc5474b19bbda50991f1ac9546b (HEAD -> humble, origin/humble)
2024-08-24T03:16:23.9790114Z Author: Stefan Fabian <fabian@sim.tu-darmstadt.de>
2024-08-24T03:16:23.9791056Z 
2024-08-24T03:16:23.9791439Z     Generate typesupport getter declarations for actions, messages and services. (#778)
2024-08-24T03:16:23.9801901Z     0.9.3
2024-08-24T03:16:23.9802289Z === src/ros2/rosidl_typesupport (git) ===
2024-08-24T03:16:23.9802961Z commit a702c179351b251403b2b188007535dac56fb33a (HEAD -> humble, origin/humble)
2024-08-24T03:16:23.9803654Z Author: Stefan Fabian <fabian@sim.tu-darmstadt.de>
2024-08-24T03:16:23.9804054Z 
2024-08-24T03:16:23.9804329Z     Added C interfaces to obtain service and action type support. (#149)
```
but
`rosdep install -r --from-paths src/ros2/rosidl/rosidl_adapter src/ros2/rosidl/rosidl_cli src/ros2/rosidl/rosidl_cmake src/ros2/rosidl/rosidl_generator_c src/ros2/rosidl/rosidl_generator_cpp src/ros2/rosidl/rosidl_parser src/ros2/rosidl/rosidl_runtime_c src/ros2/rosidl/rosidl_runtime_cpp src/ros2/rosidl/rosidl_typesupport_interface src/ros2/rosidl/rosidl_typesupport_introspection_c src/ros2/rosidl/rosidl_typesupport_introspection_cpp src/ros2/rosidl_defaults/rosidl_default_generators src/ros2/rosidl_defaults/rosidl_default_runtime src/ros2/rosidl_python/rosidl_generator_py src/ros2/rosidl_typesupport/rosidl_typesupport_c src/ros2/rosidl_typesupport/rosidl_typesupport_cpp src/ros2/rosidl_typesupport_fastrtps/fastrtps_cmake_module src/ros2/rosidl_typesupport_fastrtps/rosidl_typesupport_fastrtps_c src/ros2/rosidl_typesupport_fastrtps/rosidl_typesupport_fastrtps_cpp  --ignore-src --skip-keys 'rti-connext-dds-5.3.1 rti-connext-dds-6.0.1 ' --rosdistro humble -y`

installs ros-humble-rosidl-typesupport-cpp and others, and colcon does not build it despite being in the source folder?

May it be that there is a dependency missing in the manifest? Checking out [rosidl_typesupport](https://github.com/ros2/rosidl_typesupport/tree/humble) and [rosidl](https://github.com/ros2/rosidl/tree/humble) in my humble docker and running `colcon build --packages-up-to rosidl_generator_cpp` does not build `rosidl_typesupport_cpp`.

It will solve itself with the next humble sync, but strange that the source build does not work here.

sloretz, 2024-09-06:
https://github.com/ros-controls/ros2_controllers/actions/runs/10713321493/job/29705162169#step:9:9696

```
  In file included from /home/runner/work/ros2_controllers/ros2_controllers/ros_ws/src/ros2/rosidl/rosidl_generator_cpp/test/test_srv_initialization.cpp:17:
  /home/runner/work/ros2_controllers/ros2_controllers/ros_ws/build/rosidl_generator_cpp/rosidl_generator_cpp/rosidl_generator_cpp/srv/basic_types.hpp:10:10: fatal error: rosidl_generator_cpp/srv/detail/basic_types__type_support.hpp: No such file or directory
     10 | #include "rosidl_generator_cpp/srv/detail/basic_types__type_support.hpp"
        |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  compilation terminated.
  gmake[2]: *** [CMakeFiles/test_srv_initialization.dir/build.make:76: CMakeFiles/test_srv_initialization.dir/test/test_srv_initialization.cpp.o] Error 1
  gmake[1]: *** [CMakeFiles/Makefile2:363: CMakeFiles/test_srv_initialization.dir/all] Error 2
  gmake: *** [Makefile:146: all] Error 2
  ---
  --- stderr: rosidl_generator_cpp
  In file included from /home/runner/work/ros2_controllers/ros2_controllers/ros_ws/src/ros2/rosidl/rosidl_generator_cpp/test/test_srv_initialization.cpp:17:
  /home/runner/work/ros2_controllers/ros2_controllers/ros_ws/build/rosidl_generator_cpp/rosidl_generator_cpp/rosidl_generator_cpp/srv/basic_types.hpp:10:10: fatal error: rosidl_generator_cpp/srv/detail/basic_types__type_support.hpp: No such file or directory
     10 | #include "rosidl_generator_cpp/srv/detail/basic_types__type_support.hpp"
        |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  compilation terminated.
  gmake[2]: *** [CMakeFiles/test_srv_initialization.dir/build.make:76: CMakeFiles/test_srv_initialization.dir/test/test_srv_initialization.cpp.o] Error 1
  gmake[1]: *** [CMakeFiles/Makefile2:363: CMakeFiles/test_srv_initialization.dir/all] Error 2
  gmake: *** [Makefile:146: all] Error 2
  ---
  Failed   <<< rosidl_generator_cpp [13.7s, exited with code 2]
  ```

The tests that are failing to build were later moved to another package: https://github.com/ros2/rosidl/pull/701 , which might offer a clue as to why the action in `ros-controls/ros2_controllers` is failing.

Since the messages are generated in the same package, it seems strange to me that the test targets are not using [rosidl_get_typesupport_target](https://github.com/ros2/rosidl/blob/humble/rosidl_cmake/cmake/rosidl_get_typesupport_target.cmake) to depend on the in-package generated messages. If that was the problem though, I'd expect that to be failing in all builds.

Ali-AliAli-Ali, 2024-09-18:
Faced the same errors when trying to build ROS2 Humble on Ubuntu 22.04 via [the guide](https://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Development-Setup.html#build-the-code-in-the-workspace). Any guesses how to fix this, please?

MikeWrock, 2024-10-09:
The source build seems broken but if you cherry pick the changes in the rosidl package that broke the build you can still get it to work:

```
cd ros2/rosidl
git cherry-pick 654d6f5658b59009147b9fad9b724919633f38fe
```

MartinSolk, 2024-11-26:
Hi is anything new on this issue, because i tried to build my workspace and i have the same issue with #include "drone_messages/msg/detail/inference__type_support.hpp" not found (because it isn't created).
I tried this cherry-pick, but no success.

Ryanf55, 2024-11-27:
I'm facing this too. I just tried rebasing our `rosidl` humble fork and now it fails to build.

MartinSolk, 2024-11-27:
I don't know why, maybe somebody has an idea but i tried this cherry-pick and then restarted my computer by chance and now it works.

clalancette, 2024-11-27:
It's not enough to just compile this repository; you also need the changes from https://github.com/ros2/rosidl_typesupport/pull/149

christophfroehlich, 2024-11-27:
> It's not enough to just compile this repository; you also need the changes from [ros2/rosidl_typesupport#149](https://github.com/ros2/rosidl_typesupport/pull/149)

In the meantime, it built again in our CI. (we always compiled it from source from the ros2.repos file).

2 days ago a [new version ](https://github.com/ros2/rosidl_typesupport/commits/humble/) was released, now it might also work for the rest of you.

Ryanf55, 2024-11-27:
> It's not enough to just compile this repository; you also need the changes from [ros2/rosidl_typesupport#149](https://github.com/ros2/rosidl_typesupport/pull/149)

Yep, this makes sense. We were using the released version of rosidl_typesupport that comes from `ros:humble` docker.
Pulling that in resolved the build error.

galbertson-hss, 2025-01-06:
I encountered this issue while attempting to build ROS2 with tracing enabled, and realized that it had to do with having already sourced a `setup.bash` file before running the top-level `colcon build` command. Make sure `$AMENT_PREFIX_PATH` is empty and try the build again.