ros2/rosidl #823: rosidl_adapter: could not import 'rosidl_typesupport_c' for package geometry_msgs

flxinxout, 2024-08-29:
## Bug report
Hello. I recently experienced troubles with navigation2 installed in a docker container. The error I encounter is when I use a custom message using geometry_msgs, for example using nav2. I saw that nav2 added changes from rodisl recently.

**Required Info:**
Here is the link to the image (ros2 humble, Ubuntu 22.04):
ghcr.io/epflxplore/rover:humble-jetson

#### Steps to reproduce issue
- Pull the image
- create a package with a custom message with a geometry_msgs
- create a node and start it
- 
#### Additional information
This error seems to appear occasionally. Sometimes it appears when the image is pulled from github, sometimes when I build it locally.

sloretz, 2024-09-06:
This one got talked about today. It seems strange that this particular error would only happen occasionally. Does it ever happen after you've run `sudo apt-get dist-upgrade`?

Luigip98, 2025-02-26:
@sloretz I don't know if I have the same troubles but I noticed that when I try to create a custom interface with geometry_msgs , inside a package with an already built library(such as a ros2 controller), even changing the name in the CMakelists inside rosidl_generate_interfaces( " "), I am not able to call the service from terminal with the error: "Could not import 'rosidl_typesupport_c' for package "package_name" . What I noticed is that looking into the install/ folder : 

`find install/ -name "*rosidl_typesupport_c*"`
 
there is something like that:
install/local/lib/python3.10/dist-packages/package_name/package_name_s__rosidl_typesupport_c..cpython-310-x86_64-linux-gnu.so

that is fine unless for the double dot ".." next to cpython.
For the other packages where there are custom interfaces and everything is fine there is only one dot "." 
As a result, if I modify the name with only one dot it works.

Moreover if I build the same custom service in a blank package it works. Why it create that file .so with a name error inside?