ros2/rosidl #835: :farmer: rosidl_adapter mypy test regression timeout in Windows

Crola1702, 2024-11-06:
## Bug report

**Required Info:**

- Operating System:
  - Windows
- Installation type:
  - Source
- Version or commit hash:
  - Rolling

#### Steps to reproduce issue

1. Run a build in [windows debug](https://ci.ros2.org/job/nightly_win_deb/) 
2. See [rclpy.test_node.xunit.missing_result](https://ci.ros2.org/job/nightly_win_deb/3271/testReport/junit/(root)/rclpy/test_node_xunit_missing_result/) test regression

#### Additional information

Reference build:
* [nightly_win_deb#3271](https://ci.ros2.org/job/nightly_win_deb/3271)

Test regression:
* [projectroot.mypy](https://ci.ros2.org/job/nightly_win_deb/3271/testReport/junit/(root)/projectroot/mypy_2/)
* [rosidl_adapter.mypy.xunit.missing_result](https://ci.ros2.org/job/nightly_win_deb/3271/testReport/junit/(root)/rosidl_adapter/mypy_xunit_missing_result/)

```
test 2
    Start 2: mypy

2: Test command: C:\Python38\python_d.exe "-u" "C:/ci/ws/install/share/ament_cmake_test/cmake/run_test.py" "C:/ci/ws/build/rosidl_adapter/test_results/rosidl_adapter/mypy.xunit.xml" "--package-name" "rosidl_adapter" "--output-file" "C:/ci/ws/build/rosidl_adapter/ament_mypy/mypy.txt" "--command" "C:/ci/ws/install/Scripts/ament_mypy.exe" "--xunit-file" "C:/ci/ws/build/rosidl_adapter/test_results/rosidl_adapter/mypy.xunit.xml"
2: Working Directory: C:/ci/ws/src/ros2/rosidl/rosidl_adapter
2: Test timeout computed to be: 60
2: -- run_test.py: invoking following command in 'C:\ci\ws\src\ros2\rosidl\rosidl_adapter':
2:  - C:/ci/ws/install/Scripts/ament_mypy.exe --xunit-file C:/ci/ws/build/rosidl_adapter/test_results/rosidl_adapter/mypy.xunit.xml
1/1 Test #2: mypy .............................***Timeout  60.21 sec
```

Probably caused by https://github.com/ros2/rosidl/pull/828

InvincibleRMC, 2024-11-07:
Hmm I wonder why mypy times out when running Windows debug but the other Windows builds.

Running `mypy` locally on my Windows machine only takes 5.2 seconds.

Output from ` Measure-Command {mypy .}`
```
Days              : 0
Hours             : 0
Minutes           : 0
Seconds           : 5
Milliseconds      : 258
Ticks             : 52583442
TotalDays         : 6.08604652777778E-05
TotalHours        : 0.00146065116666667
TotalMinutes      : 0.08763907
TotalSeconds      : 5.2583442
TotalMilliseconds : 5258.3442
```

InvincibleRMC, 2024-11-11:
For resolving this issue would it be better to revert #828 or just remove the failing mypy test since the types added can still be useful?

Crola1702, 2024-11-11:
Tagging @sloretz as PR approver