ros2/rosidl #850: Feature Request: Optional Attributes

Ryanf55, 2025-02-25:
DDS supports the [optional attribute in IDL 4.2 specification](https://www.omg.org/spec/IDL/4.2/PDF) attribute section 8.3.1.3. In C++, now that we have C++17, we can use std::optional.

A prior dicussion is here: https://discourse.ros.org/t/optional-fields-in-message/991/16

Consider adding a `@optional` specifier to rosidl that translates to the DDS optional .

Amend the ROS 2 design documentation for `Standardized Annotations` to add optional:
https://design.ros2.org/articles/idl_interface_definition.html

It's supported by:
* [FastDDS](https://fast-dds.docs.eprosima.com/en/latest/fastddsgen/dataTypes/dataTypes.html#optional-members)
* [RTI](https://community.rti.com/static/documentation/connext-dds/current/doc/manuals/connext_dds_professional/extensible_types_guide/extensible_types/Optional_Members.htm)
* [CycloneDDS](https://cyclonedds.io/content/guides/supported-idl.html#optional)
* [Zenoh](https://github.com/colinhacks/zod/issues/310#issuecomment-794533682)

alsora, 2025-03-06:
Hi, we discussed this during the ROS waffle meeting and while we agree that this would be a useful feature, we also recognize that it will require a relatively large amount of work.
It would be great to get some support from the community.

Ryanf55, 2025-03-07:
Thanks. If I end up getting time allocated at work, I'll request being assigned, but otherwise anyone is welcome.

Would you accept a merge with opt-in support of `@optional` if I got it working only with the following:
* FastDDS
* rosidl_generator_cpp

We want this feature to integrate with a non-ROS-based software. The only incompatibility is the use of optional.
Getting this working on all RMW's and all languages is unlikely to be approved.

Ryanf55, 2025-07-29:
> Hey, I'm interested in taking a stab at adding optional support at least in `rosidl_generator_cpp`. One design question I have is should I add another class for `Optional` into `rosidl_parser.defintion` or simply check whether the member has the optional annotation and update places that use `msg_type_to_cpp` to something like `member_to_cpp`. Another is would the default for `std::optional<int>` be `0` or `std::nullopt`.

Nice!

According to RTI, by default, optional members should be unset, so not default initialized. So, `std::nullopt`.

https://community.rti.com/static/documentation/connext-dds/current/doc/manuals/connext_dds_professional/extensible_types_guide/extensible_types/Optional_Members.htm

`By default optional members are unset (dds::core::optional<T>::has_value() is false).`

InvincibleRMC, 2025-08-01:
When updating the actual `.msg`, `.srv`, or `.action` files to support optional fields, would you prefer introducing a new `optional` keyword before the type, using a question mark after the type (e.g., `char?` for an optional `char`), or some other approach?

christophebedard, 2025-08-01:
I think we should do `@optional` in ROSIDL (`.msg`/`.srv`/`.action`) like OMG IDL does. OMG IDL has other `@annotations` (e.g., `@key`, `@default`, `@verbatim`) which ROSIDL could potentially also support, so `@optional` would be more IDL-y.

Ryanf55, 2025-08-01:
> I think we should do `@optional` in ROSIDL (`.msg`/`.srv`/`.action`) like OMG IDL does. OMG IDL has other `@annotations` (e.g., `@key`, `@default`, `@verbatim`) which ROSIDL could potentially also support, so `@optional` would be more IDL-y.

Agree

MiguelCompany, 2025-10-31:
@InvincibleRMC There is also https://github.com/ros2/rmw_implementation/tree/rolling/test_rmw_implementation