ros2/rosidl #869: Deprecation of codecvt warning in traits.hpp

waytoashutosh, 2025-05-14:
When compiling ROS 2 with a C++17 or newer toolchain, the line 131 in rosidl_runtime_cpp/include/rosidl_runtime_cpp/traits.hpp

```
std::wstring_convert<std::codecvt_utf8_utf16<char16_t>, char16_t> convert;
```

emits a deprecation warning for `std::codecvt_utf8_utf16` and `wstring_convert` 

How do I resolve this issue, I did some research and there are no replacements as of now.

fujitatomoya, 2025-05-23:
@waytoashutosh can you pick https://github.com/ros2/rosidl/pull/871 and try to see if this works for you?

sloretz, 2025-06-05:
🧇 I don't think we've seen this warning in CI, and C++77 is the default version What platform and compiler are you using?

vermaete, 2025-06-11:
I have the same error Yocto (at master) what is using GCC 15.1:
```
In file included from /home/jan/projects/meta-ros/build/tmp/work/cortexa76-oe-linux/pcl-ros/2.7.1-1/recipe-sysroot/opt/ros/kilted/include/geometry_msgs/geometry_msgs/msg/detail/transform_stamped__traits.hpp:18,
                 from /home/jan/projects/meta-ros/build/tmp/work/cortexa76-oe-linux/pcl-ros/2.7.1-1/recipe-sysroot/opt/ros/kilted/include/geometry_msgs/geometry_msgs/msg/transform_stamped.hpp:9,
                 from /home/jan/projects/meta-ros/build/tmp/work/cortexa76-oe-linux/pcl-ros/2.7.1-1/recipe-sysroot/opt/ros/kilted/include/tf2/tf2/buffer_core.hpp:48,
                 from /home/jan/projects/meta-ros/build/tmp/work/cortexa76-oe-linux/pcl-ros/2.7.1-1/recipe-sysroot/opt/ros/kilted/include/tf2_ros/tf2_ros/async_buffer_interface.h:39,
                 from /home/jan/projects/meta-ros/build/tmp/work/cortexa76-oe-linux/pcl-ros/2.7.1-1/recipe-sysroot/opt/ros/kilted/include/tf2_ros/tf2_ros/buffer.h:42,
                 from /home/jan/projects/meta-ros/build/tmp/work/cortexa76-oe-linux/pcl-ros/2.7.1-1/git/include/pcl_ros/transforms.hpp:41,
                 from /home/jan/projects/meta-ros/build/tmp/work/cortexa76-oe-linux/pcl-ros/2.7.1-1/git/src/transforms.cpp:37:
/home/jan/projects/meta-ros/build/tmp/work/cortexa76-oe-linux/pcl-ros/2.7.1-1/recipe-sysroot/opt/ros/kilted/include/rosidl_runtime_cpp/rosidl_runtime_cpp/traits.hpp: In function 'void rosidl_generator_traits::v\
alue_to_yaml(const std::u16string&, std::ostream&)':
/home/jan/projects/meta-ros/build/tmp/work/cortexa76-oe-linux/pcl-ros/2.7.1-1/recipe-sysroot/opt/ros/kilted/include/rosidl_runtime_cpp/rosidl_runtime_cpp/traits.hpp:132:8: warning: 'template<class _Codecvt, cla\
ss _Elem, class _Wide_alloc, class _Byte_alloc> class std::__cxx11::wstring_convert' is deprecated [-Wdeprecated-declarations]
  132 |   std::wstring_convert<std::codecvt_utf8_utf16<char16_t>, char16_t> convert;
      |        ^~~~~~~~~~~~~~~
In file included from /home/jan/projects/meta-ros/build/tmp/work/cortexa76-oe-linux/pcl-ros/2.7.1-1/recipe-sysroot/usr/include/c++/15.1.0/locale:47,
                 from /home/jan/projects/meta-ros/build/tmp/work/cortexa76-oe-linux/pcl-ros/2.7.1-1/recipe-sysroot/usr/include/c++/15.1.0/iomanip:53,
                 from /home/jan/projects/meta-ros/build/tmp/work/cortexa76-oe-linux/pcl-ros/2.7.1-1/recipe-sysroot/opt/ros/kilted/include/rosidl_runtime_cpp/rosidl_runtime_cpp/traits.hpp:20:
/home/jan/projects/meta-ros/build/tmp/work/cortexa76-oe-linux/pcl-ros/2.7.1-1/recipe-sysroot/usr/include/c++/15.1.0/bits/locale_conv.h:262:33: note: declared here
  262 |     class _GLIBCXX17_DEPRECATED wstring_convert
      |                                 ^~~~~~~~~~~~~~~
```